ros2/launch #376: Enable var substitution inside node XML args attribute string

ruffsl, 2020-02-04:
Var substitution is not preformed for var references in the args attribute string for node XML elements when using launch-xml. This was possible in ROS1, but is not yet supported in ROS2, thus blocking the port of ROS1 launch files where nodes expect a numerous run time arguments set from the XML launch file configuration. E.g. when ros nodes wrap around libraries that are initialized via arbitrary GFlags and GLog configurations via run time command arguments.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros-eloquent-launch-xml/bionic,now 0.9.6-1bionic.20200121.220435 amd64

#### Steps to reproduce issue

Example:

``` xml
<launch>
  <arg name="dataset_name" default="foo"/>
  <arg name="online" default="true" />
  <arg name="log_output" default="false"/>
  <arg name="log_output_path" default="$(find-pkg-share vio)/output_logs/"/>
  <arg name="use_lcd" default="false"/>
  <arg name="params_folder" default="$(find-pkg-share vio)/param/$(var dataset_name)"/>
  <arg name="path_to_vocab" default="$(find-pkg-share vio)/vocabulary/ORBvoc.yml"/>
  <arg name="verbosity"    default="0" />
  <arg name="backend_type" default="1" />

   <node pkg="some_ros_gflag_package" exec="some_ros_gflag_node"
        args="--use_lcd=$(var use_lcd) \
            --lcd_params_path=$(var params_folder)/LCDParameters.yaml \
            --vocabulary_path=$(var path_to_vocab) \
            --flagfile=$(var params_folder)/flags/Mesher.flags \
            --flagfile=$(var params_folder)/flags/VioBackEnd.flags \
            --flagfile=$(var params_folder)/flags/RegularVioBackEnd.flags \
            --flagfile=$(var params_folder)/flags/Visualizer3D.flags \
            --v=$(var verbosity)
            --backend_type=$(var backend_type)
            --log_output=$(var log_output)
            --output_path=$(var log_output_path)">/>
</launch>
```

#### Expected behavior

Running `ros2 launch foo.launch.xml` results in passing the expanded args string with respective substations onto the executable.

#### Actual behavior

Running `ros2 launch foo.launch.xml` results in passing the partial args string with respective substations being blank or empty strings.

ruffsl, 2020-02-04:
On closer inspection, it looks like launch-xml is substituting the vars; only that it seems that args are prefixed and postfixed by unspecified spaces, e.g:

``` diff
<arg name="foo" default="bar"/>
...
- spam=$(var foo)/baz.yml
+ spam= bar /baz.yml
```

jacobperron, 2020-02-05:
I think this is a duplicate of https://github.com/ros2/launch/issues/337

ivanpauno, 2020-02-06:
Closing as duplicate of #337.

Just to clarify: The problem particularly happens when using `args` attribute of `node`, and not everywhere.

closed 2020-02-06 by ivanpauno