ros2/launch #377: Command substitution: special characters inside XML comments causes launch to fail

jacobperron, 2020-02-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from source 
- Version or commit hash:
  - 4333910ba41977f3ab9867082c86a8698ada3a16 

Specifically, I've noticed that colons (`:`) and tab characters are not working.

#### Steps to reproduce issue

I ran into this issue trying to pass a URDF from xacro to robot_state_publisher, like the [example here](https://github.com/ros/robot_state_publisher/blob/f877cbf4399d8c497b946376664bd76b0c192920/launch/rsp-launch-xacro-command-subst.xml). For some reason, if the URDF/xacro file contains a colon (`:`) inside of an XML comment, then launch fails.

Here's my example URDF file `foo.urdf`:

```xml
<?xml version="1.0" ?>
<robot>
  <!--: -->
</robot>
```

And my launch file:

```xml
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
    <param name="foo" value="$(command 'cat foo.urdf')" />
  </node>
</launch>
```

#### Expected behavior

The launch file succeeds and we see the talker node chatting.

#### Actual behavior

Error:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): mapping values are not allowed here
  in "<unicode string>", line 3, column 7:
      <!--: -->
          ^
```

#### Additional information

The error seems to specifically occur if there is a colon followed by a space. E.g. making the following change works fine:

```diff
-      <!--: -->
+      <!--:-->
```

Replacing the colcon with a tab character also triggers the same error:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): while scanning for the next token
found character '\t' that cannot start any token
  in "<unicode string>", line 3, column 7:
      <!--	-->
          ^
```

ivanpauno, 2020-02-05:
That seems extremly weird.
Can you run launch in debug mode (`-d`) and share the traceback?

hidmic, 2020-02-05:
Wild guess: we're YAML parsing command output. Try adding quotes before and after the command substitution.

ivanpauno, 2020-02-05:
> Wild guess: we're YAML parsing command output. Try adding quotes before and after the command substitution.

Ouch! Yes, that's true, parameters values that are strings are parsed like yaml.
There's an issue about that https://github.com/ros2/launch_ros/issues/74 (I think I opened an issue proposing typed substitutions, but I haven't found it).
At the moment, I agree with Mich that the best solution is adding extra quotes.

jacobperron, 2020-02-05:
Do you mean like this?

```xml
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
    <param name="foo" value="'$(command 'cat foo.urdf')'" />
  </node>
</launch>
```

I get the same error.

jacobperron, 2020-02-05:
Here's an example output with --debug:

```
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/jacob/.ros/log/2020-02-05-14-06-28-231429-warner-24708
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f6fac362630>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f6fac362630>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f6fac380fd0>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f6fac362748>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f6fac362748>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f6fac380fd0>'
[DEBUG] [launch]: Traceback (most recent call last):
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/launch_service.py", line 385, in run_async
    await process_one_event_task
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 126, in send
    return self.gen.send(value)
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/launch_service.py", line 295, in _process_one_event
    await self.__process_event(next_event)
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/launch_service.py", line 315, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/home/jacob/ws/latest_ws/install/launch_ros/lib/python3.6/site-packages/launch_ros/actions/node.py", line 350, in execute
    self._perform_substitutions(context)
  File "/home/jacob/ws/latest_ws/install/launch_ros/lib/python3.6/site-packages/launch_ros/actions/node.py", line 321, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/home/jacob/ws/latest_ws/install/launch_ros/lib/python3.6/site-packages/launch_ros/utilities/evaluate_parameters.py", line 145, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/home/jacob/ws/latest_ws/install/launch_ros/lib/python3.6/site-packages/launch_ros/utilities/evaluate_parameters.py", line 70, in evaluate_parameter_dict
    yaml_evaluated_value = yaml.safe_load(evaluated_value)
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/constructor.py", line 41, in get_single_data
    node = self.get_single_node()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/composer.py", line 58, in compose_document
    self.get_event()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/parser.py", line 118, in get_event
    self.current_event = self.state()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/parser.py", line 193, in parse_document_end
    token = self.peek_token()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/scanner.py", line 129, in peek_token
    self.fetch_more_tokens()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/home/jacob/.local/lib/python3.6/site-packages/yaml/scanner.py", line 579, in fetch_value
    self.get_mark())
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 3, column 7:
      <!--: -->
          ^

[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f6fac38f198>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f6fa8039b70>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f6fac38f198>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f6fac3629e8>'
Executing <Task finished coro=<LaunchService._process_one_event() done, defined at /home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/launch_service.py:293> result=None created at /home/jacob/ws/latest_ws/install/launch/lib/python3.6/site-packages/launch/launch_service.py:363> took 0.994 seconds
```

ivanpauno, 2020-02-06:
> Do you mean like this?

TL;DR no, you need duplicated single quotes :man_facepalming: 

```xml
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
    <param name="foo" value="''$(command 'cat foo.urdf')''" />
  </node>
</launch>
```

why? well, each value read from an xml file is coerced with some rules to float/int/bool/str.
One of the rules, it's that outer single quotes are implicit in an string. If you find explicit quotes, they are dropped (similar of what yaml does).
After that, the `Node` action itself is parsing the string like yaml ....

We should definitly fix this, before the problem gets proliferated in multiple launch files.
I think that the easier fix, it's to specify the type of the parameter (somehow), instead of using yaml rules in the implementation of `Node` action (see https://github.com/ros2/launch_ros/issues/74#issuecomment-533257827).

ddengster, 2020-02-12:
```
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
      <param name="foo" value="$(command 'pwd')" />
  </node>
</launch>
```

So I'm facing something similar to this issue. Has $(command '..') been deployed for #375 ?  It seems anything I'm trying with $(command '..') doesn't work. I've built ros2/launch within my colcon workspace and confirmed that ros2 launch commands runs with what I've built.

ivanpauno, 2020-02-13:
Currently, the following should work:
```xml
<launch>
  <node pkg="demo_nodes_cpp" exec="talker" output="screen">
      <param name="foo" value="''$(command 'pwd')''" />
  </node>
</launch>
```

I will work on a fix so that's no more needed.

mogumbo, 2020-03-09:
I just got back to looking at this. It almost works for me. I'm doing:
`
<node name="robot_state_publisher" pkg="robot_state_publisher" exec="robot_state_publisher">
    <param name="robot_description" value="''$(command '$(find-pkg-prefix xacro)/bin/xacro
                                           $(find-pkg-share krex2_description)/urdf/rp1a_robot.xacro')''"/>
  </node>
`
Which gives me this output:
`
[ERROR] [launch]: Caught exception in launch (see debug for traceback): expected '<document start>', but found '<scalar>'
  in "<unicode string>", line 96, column 11:
            I'm not sure if just specifying tw ... 
              ^
`
If I remove all the single-quotes from the comments in my xacro file, I can launch my robot. Any thoughts on what can be done about this?

ivanpauno, 2020-03-10:
> Which gives me this output:
> 
> ```shell
> [ERROR] [launch]: Caught exception in launch (see debug for traceback): expected '<document start>', but found '<scalar>'
>   in "<unicode string>", line 96, column 11:
>             I'm not sure if just specifying tw ... 
>               ^
> ```

:man_facepalming: That's a fun one.
 
> If I remove all the single-quotes from the comments in my xacro file, I can launch my robot. Any thoughts on what can be done about this?

I will describe first the problem. The value of `Node` action [parameters](https://github.com/ros2/launch_ros/blob/be0f9a1755df7526cff590a7027054501b5128c3/launch_ros/launch_ros/actions/node.py#L71) argument are read like `yaml`. That was proposed originally in https://github.com/ros2/launch_ros/pull/31, because substitutions are only resolved to a `str`.
https://github.com/ros2/launch_ros/pull/33 limited in which cases a parameter value was read like yaml (only if the original provided value was a substitution, avoided when it was a explicitly passed string).
To solve the problem, my idea was to stop automatically doing the yaml load, and provide an explicit way of specifying the parameter type (one of the options could be to use `yaml` to infer the type).

ivanpauno, 2020-03-10:
> To solve the problem, my idea was to stop automatically doing the yaml load, and provide an explicit way of specifying the parameter type (one of the options could be to use yaml to infer the type).

I've actually started to work in a fix , though I haven't had time to finish it.

ivanpauno, 2020-03-24:
Moving to `launch_ros`, as the origin of the problem is that `parameters` `Node` argument is implicitly loading passed values like `yaml`.