ros2/launch #382: ExecuteProcess cmd parameter treatment 'The passed value needs to be a dictionary in YAML format'

ddengster, 2020-02-17:
## Bug report

- Operating System:
Ubuntu 18.04
- Installation type:
Eloquent
- Version or commit hash:
#379 , but should occur in the master branch #375 

#### Steps to reproduce issue
ros2 launch mypkgname mylaunchfile.launch.py

```

import launch
import launch.actions 
import launch.substitutions 
import launch_ros.actions
import os

from launch_ros.actions import Node

def generate_launch_description():
  
  return launch.LaunchDescription([
    launch.actions.ExecuteProcess(
      name='fake_joint_calibration',
      cmd = ['ros2', 'topic', 'pub', '/calibrated', 'std_msgs/Bool', '"data: 1"'],
      output = 'screen',
      shell='True'
      )
  ])

```

#### Expected behavior
The same as running the following on the cmdline
> ros2 topic pub /calibrated std_msgs/Bool "data: 1"

Output:
```
publisher: beginning loop
publishing #1: std_msgs.msg.Bool(data=True)

publishing #2: std_msgs.msg.Bool(data=True)

...
```
#### Actual behavior
Output gives:
```
[fake_joint_calibration-1] The passed value needs to be a dictionary in YAML format
```

#### Additional information

I'd also like to know how the cmd parameters work; I've had other related weirdness with my parameters being treated like yaml. Why is that so and is there a workaround for this?

ivanpauno, 2020-02-17:
`ros2 topic pub` expects the value to be a dictionary in yaml format:

```python3

def generate_launch_description():
  
  return launch.LaunchDescription([
    launch.actions.ExecuteProcess(
      name='fake_joint_calibration',
      cmd = ['ros2', 'topic', 'pub', '/calibrated', 'std_msgs/Bool', '"{data: True}"'],
      output = 'screen',
      shell='True'
      )
  ])
```

That should work.

Closing, as it's unrelated to launch.

closed 2020-02-17 by ivanpauno

jeremy-shannon, 2023-02-08:
Hi, I know this topic is closed but I was having the same problem and this is the only place I found anything about it whatsoever. I think it's worth noting for anyone else who might come along that I tried your suggestion:

```
def generate_launch_description():
  
  return launch.LaunchDescription([
    launch.actions.ExecuteProcess(
      name='fake_joint_calibration',
      cmd = ['ros2', 'topic', 'pub', '/calibrated', 'std_msgs/Bool', '"{data: True}"'],
      output = 'screen',
      shell='True'
      )
  ])
```
...and still got the 'The passed value needs to be a dictionary in YAML format' error. But changing the 'cmd' line to the following (i.e. removing the double quotes) ended up working:

`cmd = ['ros2', 'topic', 'pub', '/calibrated', 'std_msgs/Bool', '{data: True}'],`

asymingt, 2023-05-31:
Thank you @jeremy-shannon -- I spent about an hour trying to work this out, trying various combinations of `yaml.dump()`, single and double quotations! It was your observation that helped get my code working.

jeremy-shannon, 2023-06-01:
@asymingt 
> Thank you @jeremy-shannon -- I spent about an hour trying to work this out, trying various combinations of `yaml.dump()`, single and double quotations! It was your observation that helped get my code working.

Nice! I’m glad to hear it helped. 😁