ros2/launch #383: Support setting log-level via command-line

norro, 2020-02-18:
## Feature request

#### Feature description
Allow setting log-level via command line, similar as documented for `ros2 run`, `ros2 service`, etc. see [ros2/Tutorials/Logging-and-logger-configuration](https://index.ros.org/doc/ros2/Tutorials/Logging-and-logger-configuration/)

### Steps to reproduce issue
```
ros2 launch demo_nodes_cpp talker_listener.launch.py --ros-args --log-level debug
```

#### Expected behavior
log level is set to debug

#### Actual behavior
`ros2: error: unrecognized arguments: --ros-args --log-level debug`

claireyywang, 2020-02-27:
@norro Would you be interested in supplying a pull request with this feature?

claireyywang, 2020-02-27:
@norro Could you provide more information about why this feature is needed and what it would do?

tim-fan, 2020-10-22:
Hello, I am also interested in this feature, for my use case I'd like to sometimes launch my system at logging level WARN, and sometimes at logging level ERROR (basically I'd like to easily switch between more or less verbose options).

hidmic, 2020-10-22:
First of all, it's important not to mix up ROS 2 node arguments with ROS 2 cli arguments. `--ros-args` applies to ROS 2 nodes **only**. You may pass it to `ros2 run` solely because it parses none of it and forwards it all to the underlying process.

Having said that, currently this can be achieved by explicitly declaring a launch argument and then using that argument to set `launch_ros.actions.Node` argument. Something along the lines of:

```python
def generate_launch_description():
      return launch.LaunchDescription([
          launch.actions.DeclareArgument(name='log_level', default_value='info'),
          launch_ros.actions.Node(
                # name, namespace, executable, etc.
                arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level')]
          )
      ])
```

should do. 

It _could_ be made implicit and applicable to all nodes, though in that case I wonder if it's worth all the plumbing instead of simply having `rcl` look at some environment variable to select the default logger level. Would something like that satisfy your use cases @norro @tim-fan? CC @wjwwood for valuable feedback.

norro, 2020-10-26:
That would satisfy my needs and would even provide a more fine-grained handle than I initially envisioned.
As this is currently already possible, I would close this issue if @tim-fan doesn't object.

Thanks for the great suggestion, @hidmic!

tim-fan, 2020-10-26:
Hi, thanks for the feedback.

I had considered the suggestion above of passing log-level arguments down to individual nodes. An issue with this for me is when I am including other launch files from external packages, in which case I cannot add the extra arguments. I'd say it may get a bit unwieldy in general for larger projects, if every launch file needs to pass logging levels to every node, in order to achieve this global level of control over logging.

As an alternative, the "environment variable sets default logging level" suggestion sounds good to me 👍

Aposhian, 2021-06-23:
This applies equally to [name remapping and parameter overrides](https://docs.ros.org/en/foxy/Guides/Node-arguments.html), which are also often done at runtime (desirable as a command line option). It would be very convenient to forward these options to all of the nodes, since these are things that are good candidates to change at runtime, and it is burdensome to have to edit the launch code to do so, even to add boilerplate to forward these options. There would be some edge cases that wouldn't make a ton of sense, especially for name remapping, which can be node specific like `__node:=new_node_name`. However, we could just say that the expected behavior in that case is just giving all the nodes the same name, and if that is a bug, then it is your fault for specifying that command line option.

saahu27, 2022-11-08:
This way, you can set log level to nodes. Try this launch file. This worked for me.

```
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
import launch_ros.actions
from launch.substitutions import TextSubstitution
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    return LaunchDescription([

        DeclareLaunchArgument(
                "Parameter_launch_argument", default_value=TextSubstitution(text=str("Parameter_launch")),
                description="Parameter launch default value"
            ),

        DeclareLaunchArgument(
            "log_level",
            default_value = TextSubstitution(text=str("WARN")),
            description="Logging level"
        ),

        launch_ros.actions.Node(
            package="serverpublisher",
            executable="talker",
            name="custom_msg_publisher_to_Life_iteration_topic",
            parameters=[
                {"Parameter_Publisher":  LaunchConfiguration('Parameter_launch_argument')}
            ],
            output="screen",
            arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level')]
        ),

        launch_ros.actions.Node(
            package="serverpublisher",
            executable="server",
            name="custom_server_to_add_three_ints",
            output="screen",
            arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level')]
        ),

        launch_ros.actions.Node(
            package="serverpublisher",
            executable="talkerandlistner",
            name="custom_msg_subscriber_to_No_Life_iteration_topic",
            output="screen",
            arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level')]
        )
    ])
```

artemiialessandrini, 2022-11-24:
Coming from Galactic, node logging has been [introduced](https://docs.ros.org/en/foxy/Releases/Release-Galactic-Geochelone.html#ability-to-specify-per-logger-log-levels), which sets up log level to specific node, avoiding global log level

From the last example, it looks like this:
``` sh   
        <...>
        DeclareLaunchArgument(
            "log_level",
            default_value = TextSubstitution(text=str("WARN")),
            description="Logging level"
        ),

        launch_ros.actions.Node(
            package="serverpublisher",
            executable="talker",
            name="custom_msg_publisher_to_Life_iteration_topic",
            parameters=[
                {"Parameter_Publisher":  LaunchConfiguration('Parameter_launch_argument')}
            ],
            output="screen",
            arguments=['--ros-args', '--log-level', ['custom_msg_publisher_to_Life_iteration_topic:=', LaunchConfiguration('log_level')]]
        ), # change here

        launch_ros.actions.Node(
            package="serverpublisher",
            executable="server",
            name="custom_server_to_add_three_ints",
            output="screen",
            arguments=['--ros-args', '--log-level', ['custom_server_to_add_three_ints:=', LaunchConfiguration('log_level')]]
        ), # change here
```

Slightly upsetting, from my understanding you **cannot** pass the Node `name` making it more elegant:
```  arguments=['--ros-args', '--log-level', [name, ':=', LaunchConfiguration('log_level')] ```