ros2/launch #387: DeprecationWarning: The loop argument is deprecated since Python 3.8

dirk-thomas, 2020-02-19:
And scheduled for removal in Python 3.10.

E.g. https://github.com/ros2/launch/blob/f243fc6b3bcd276c194151e6933cda90761b1944/launch/launch/actions/timer_action.py#L84

dirk-thomas, 2020-03-10:
Instead of creating separate tickets for each deprecation warning I will add more here:

* `launch_testing/test/launch_testing/test_launch_test_runner_validation.py:15`:

  > DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses

zmk5, 2020-04-27:
I submitted a fix for the second deprecation warning. I'm still trying to figure out what the deprecation in the first one is.

zmk5, 2020-04-27:
Just checked. It actually didn't work for me. I'll try and fix the issue.

zmk5, 2020-04-30:
Hey @dirk-thomas I have resolved all deprecation warnings associated with this issue. The team has also reviewed and merged the changes. You can close this issue if you would like 😄

ivanpauno, 2020-04-30:
Thanks for your contribution @zmk5 !!!

closed 2020-04-30 by ivanpauno