ros2/launch #392: Provide roslaunch xml schema definition downloadable

adrienbarral, 2020-03-20:
It would be nice to have the roslaunch xml definition (https://design.ros2.org/articles/roslaunch_xml.html#let-tag) downloadable somewhere.

I guess that the good place should be here : http://download.ros.org/schema/ (close to the package format definition).

ivanpauno, 2020-06-26:
Hi @adrienbarral @joshwapohlmann !

I think that adding schema validation is a good idea.
We could provide an online resource, or maybe we could also provide the file with the installation of launch/launch_ros.

To make things simple, we could move the schema definition to `launch_ros` repository, and it will be available in a link similar to:

https://raw.githubusercontent.com/ros2/design/gh-pages/articles/specs/launch.0.1.1.xsd

Does that sound like a good idea?

Some extra notes:

- It should be added to [launch_ros](https://github.com/ros2/launch_ros) repo and not here, as launch_ros adds extra functionality. Ideally, we should have two schemas: one with bare `launch` functionality, and one in top of that with the extra `launch_ros` features.
- We should use the schema in some of the tests here and in launch_ros repo, to make sure we keep it up to date.
- The main reason to move the schema here from [ros2/design](https://github.com/ros2/design/blob/gh-pages/articles/specs/launch.0.1.1.xsd) is that we could easily update implementation and schema in one PR.
- If `launch_xml` finds a launchfile that provides a schema link, we should make sure it makes the schema validation.
- We could potentially have schema validation for launch_yaml files (e.g. using [pyrx](https://github.com/uniphil/pyrx)).

adrienbarral, 2020-06-26:
Yes it sound like a very good idea.

I am ros user, so I don't really care of a "launch only" schema definition, but that can make sense to have two schemas.

ivanpauno, 2020-06-26:
> Yes it sound like a very good idea.
> 
> I am ros user, so I don't really care of a "launch only" schema definition, but that can make sense to have two schemas.

Yes, starting only with `launch_ros` sounds good.

There's no plan to address this issue in a near future, so contributions to address some of the bullets above are more than welcomed.

Ryanf55, 2024-01-26:
ament_lint_xmllint could then automatically validate launch files in your repo are compliant to the schema.

ardabbour, 2024-08-15:
I don't know if this helps, but someone did it here: [https://gist.github.com/SubaruArai/957b7e2ebe212eb2a07d26a73f9a31c3](https://gist.github.com/SubaruArai/957b7e2ebe212eb2a07d26a73f9a31c3)

So it can be used like this:

```xml
<?xml version="1.0"?>
<?xml-model href="https://gist.githubusercontent.com/nalt/dfa2abc9d2e3ae4feb82ca5608090387/raw/roslaunch.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<launch>
  <!-- Stuff -->
</launch>
```

Of course, it would be great if it were hosted on a url such as [http://download.ros.org/schema/launch.xsd](http://download.ros.org/schema/launch.xsd)

peci1, 2025-03-30:
If a launch_ros-specific schema is used, is there a way to provide additional launch actions to the validator?