ros2/launch #394: Passing a string to Node arguments behaves unintuitively

rotu, 2020-03-24:
In Node constructor, arguments takes an arguments of list of strings. Passing a string to arguments typechecks but provides unexpected results, treating the string as a sequence of single-character arguments.

e.g the following code:
```
    driver_node = LifecycleNode(package='ros2_ouster',
                                node_executable='ouster_driver',
                                node_name=node_name,
                                output='screen',
                                emulate_tty=True,
                                parameters=[parameter_file],
                                node_namespace='/',
                                arguments='__log_level:=debug'
                                )
```

This displays no error message but the mistake is visible when the process dies.
```
[ERROR] [ouster_driver-1]: process has died [pid 309341, exit code -15, cmd '/opt/ros/master/install/lib/ros2_ouster/ouster_driver _ _ l o g _ l e v e l : = d e b u g --ros-args -r __node:=ouster_driver --params-file /opt/ros/master/install/share/ros2_ouster/params/os1.yaml -r __ns:=/'].
```

Passing a string to `arguments` (or similar places) should either:

1. Split the string on spaces
2. Alert the user (`ArgumentError` seems apt)

ivanpauno, 2020-03-24:
`cmd` is expected to be a list of `Substitutions`, https://github.com/ros2/launch/blob/ae12701630c7acd3d78806bc764bf96b54d6ebe8/launch/launch/actions/execute_process.py#L90

where a substitution can be the following:
https://github.com/ros2/launch/blob/ae12701630c7acd3d78806bc764bf96b54d6ebe8/launch/launch/some_substitutions_type.py#L24-L28

Strings are iterable in `Python`, so this will think that each letter in the string is an argument of the command:
https://github.com/ros2/launch/blob/ae12701630c7acd3d78806bc764bf96b54d6ebe8/launch/launch/actions/execute_process.py#L203

I think we should detect if the input is an string, and raise an error.

ivanpauno, 2020-03-24:
This is related too: https://github.com/ros2/launch/issues/263.

rotu, 2020-03-24:
I think you're right that it should raise an error. It seems wrong to have `ExecuteProcess(cmd='echo', arguments='hello          world')` print 'hello world' or to have `ExecuteProcess(cmd='ls', arguments='-l -A')` crash, so an error seems sensible.