ros2/launch #396: Roslaunch XML Schema Fails Validation

comkieffer, 2020-04-08:
# Bug report

- Operating System: N/A
- Installation type: N/A
- Version or commit hash: N/A  
- DDS implementation: N/A
- Client library (if applicable): N/A

#### Steps to reproduce issue

Manually download [xml schema](https://design.ros2.org/articles/roslaunch_xml.html) and save it to `roslaunch_schema.xsd`. Validate a launch file with `xmllint`

``` 
echo "<launch></launch>" | xmllint --schema roslaunch_schema.xsd --noout -
```

Alternatively, use [online validator](http://www.utilities-online.info/xsdvalidation/?save=48c514a7-5f7f-4048-a7b0-d74f47c58f43-xsdvalidation) (press `Check XSD Validity` to see errors)

#### Expected behavior
Schema validation should be successful. 

#### Actual behavior

Schema validation fails in both `xmllint` (based on libxml2) and online validator (using Java libraries).

xmllint output is: 

```roslaunch_xml_schema.xsd:11: element element: Schemas parser error : Element '{http://www.w3.org/2001/XMLSchema}element': The attribute 'maxOccurs' is not allowed.
roslaunch_xml_schema.xsd:18: element choice: Schemas parser error : Element '{http://www.w3.org/2001/XMLSchema}element': The content is not valid. Expected is (annotation?, ((simpleType | complexType)?, (unique | key | keyref)*)).
roslaunch_xml_schema.xsd:143: element sequence: Schemas parser error : Element '{http://www.w3.org/2001/XMLSchema}complexType': The content is not valid. Expected is (annotation?, (simpleContent | complexContent | ((group | all | choice | sequence)?, ((attribute | attributeGroup)*, anyAttribute?)))).
roslaunch_xml_schema.xsd:478: element assert: Schemas parser error : Element '{http://www.w3.org/2001/XMLSchema}complexType': The content is not valid. Expected is (annotation?, (simpleContent | complexContent | ((group | all | choice | sequence)?, ((attribute | attributeGroup)*, anyAttribute?)))).
roslaunch_xml_schema.xsd:528: element attribute: Schemas parser error : Element '{http://www.w3.org/2001/XMLSchema}element': The content is not valid. Expected is (annotation?, ((simpleType | complexType)?, (unique | key | keyref)*)).
WXS schema roslaunch_xml_schema.xsd failed to compile
```

hidmic, 2020-04-16:
@comkieffer thank you for reporting. See https://github.com/ros2/design/pull/278. 

Unfortunately, `xmllint` will not be able to validate `launch`'s schema, as AFAIK the former can only handle XSD 1.0 and the latter is XSD 1.1

comkieffer, 2020-04-27:
Thanks  for pointing me towards the bug report. Nice to see that somebody's on it.

closed 2020-04-27 by comkieffer