ros2/launch #397: New flake8 F821 error in launch.

nuclearsandwich, 2020-04-10:
## Bug report

**Required Info:**

- Operating System:
  - cross-platform
- Installation type:
  - source
- Version or commit hash:
  - https://github.com/ros2/launch/commit/0ce19ac6b9e02e93d58ba4ddfd31e3a2d3fac36f (HEAD)
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Example build [linux nightly release](https://ci.ros2.org/view/nightly/job/nightly_linux_release/1508/testReport/junit/launch.src.ros2.launch.launch.test/test_flake8/test_flake8/)

#### Additional information

With pyflakes 2.2.0 the line below is triggering the flake8 error
F821 undefined name 'Action'
https://github.com/ros2/launch/blob/0ce19ac6b9e02e93d58ba4ddfd31e3a2d3fac36f/launch/launch/event_handlers/on_execution_complete.py#L42

I wasn't familiar enough with the typing expressions and paged @mjcarroll for a second set of eyes. From that conversation the code as is should be a valid ForwardRef but its unclear if it needs to be a ForwardRef or if we should just import the type at the file header and use it. I think he has more context to add as well.

nuclearsandwich, 2020-04-10:
@hidmic and @jacobperron are contributors on this file who might have info on why the forwardref is preferred.

mjcarroll, 2020-04-10:
My best guess is that it's related to this which landed in 2.2.0: https://github.com/PyCQA/pyflakes/pull/479  It fixes a similar issue, but may have broken this case here?

I'm not convinced that this isn't a completely valid type specification, but it may the that `pyflakes` lacks suitable interpreter context to get it right.

wjwwood, 2020-04-10:
I can fix it I think. Making a pr now.

closed 2020-04-10 by wjwwood