ros2/launch #399: [Eloquent] XML var substitutions appending an extra whitespace token

allenh1, 2020-04-10:
## Bug report

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - latest
- DDS implementation:
  - n/a
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Execute this launch file, for example.

```xml
<launch>
  <arg name="sensor" default="sensor"/>
  <arg name="sensor_link" default="front_fwd_sensor_link"/>
  <arg name="sensor_link_tf" default="0 0.1 1.0 0 0 1.5708"/>

  <node pkg="tf2_ros"
       exec="static_transform_publisher"
       name="$(var sensor)_tf_to_$(var sensor_link)"
       args="$(var sensor_link_tf) $(var sensor_link) $(var sensor)_link"/>
</launch>
```

#### Expected behavior

The `static_transform_publisher` node should launch.

#### Actual behavior

The node fails to execute due to the incorrect number of arguments.

```
allenh1@hunter-laptop:~$ ros2 launch ~/test.launch.xml
[INFO] [launch]: All log files can be found below /home/allenh1/.ros/log/2020-04-10-15-42-08-008729-hunter-laptop-10470
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sensor_tf_to_front_fwd_sensor_link-1]: process started with pid [10497]
[sensor_tf_to_front_fwd_sensor_link-1] [ERROR] []: static_transform_publisher exited due to not having the right number of arguments
[ERROR] [sensor_tf_to_front_fwd_sensor_link-1]: process has died [pid 10497, exit code 255, cmd '/opt/ros/eloquent/lib/tf2_ros/static_transform_publisher 0 0.1 1.0 0 0 1.5708 front_fwd_sensor_link sensor _link --ros-args'].
``` 

#### Additional information

I suppose this is because `$(var sensor)_link` somehow becomes `sensor _link` (note the extra ' '). I'm not sure why it fails here, but succeeds in the name of the node itself.

I haven't tested this with my source install, but I can try that later this evening if that will give any extra insight.

allenh1, 2020-04-16:
> I think this has been fixed for Foxy in #337.

That would be great news! I will test this on my source installation next time I'm on it.

> would it be possible to backport it?

That would be even better! Since this is used in a lot of launch files I use, I would highly recommend it be backported.

ivanpauno, 2020-04-16:
Backport merged, closing!

closed 2020-04-16 by ivanpauno

allenh1, 2020-04-17:
> (edit) the change wasn't released yet. It will be part of next Eloquent patch release.

Good to know -- thanks for that!