ros2/launch #400: [Launch Testing] failure messages are generic, unuseful

rotu, 2020-04-11:
In ros2cli, there are tests that command line tools produce expected output. e.g. https://github.com/ros2/ros2cli/blob/96cf5b949d1aa870111bba0664eca1c359d3307c/ros2lifecycle/test/test_cli.py#L363-L374

notably this command which uses `expect_output`:
```
        assert launch_testing.tools.expect_output(
            expected_lines=['unconfigured [1]'],
            text=lifecycle_command.output,
            strict=True
        )
```

When these tests fail, they are reported in CI with an extremely generic error message, stating only the test case and its parameter:

https://ci.ros2.org/view/nightly/job/nightly_osx_debug/1588/testReport/ros2lifecycle.src.ros2.ros2cli.ros2lifecycle.test/test_cli/test_cli/

> ## Error Message
> some test cases have failed: test_cli[rmw_cyclonedds_cpp] failed at TestROS2LifecycleCLI.test_list_all_lifecycle_hidden_node_transitions_with_hidden_flag[rmw_cyclonedds_...
> test_cli[rmw_cyclonedds_cpp] failed at TestROS2LifecycleCLI.test_list_all_lifecycle_node_transitions[rmw_cyclonedds_cpp]
> 
> ## Stacktrace
> some test cases have failed:
> test_cli[rmw_cyclonedds_cpp] failed at TestROS2LifecycleCLI.test_list_all_lifecycle_hidden_node_transitions_with_hidden_flag[rmw_cyclonedds_cpp]
> ...
> test_cli[rmw_cyclonedds_cpp] failed at TestROS2LifecycleCLI.test_list_all_lifecycle_node_transitions[rmw_cyclonedds_cpp]

rotu, 2020-04-11:
@dirk-thomas I notice more verbose error output is present in the full text log:
```
----------------------------- Captured stderr call -----------------------------
test_count_all_lifecycle_nodes[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_count_lifecycle_nodes[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_get_hidden_lifecycle_node_state[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_get_lifecycle_node_state[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_get_nonexistent_lifecycle_node_state[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_list_all_lifecycle_hidden_node_transitions_with_hidden_flag[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_list_all_lifecycle_hidden_node_transitions_without_hidden_flag[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_list_all_lifecycle_node_transitions[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_list_all_lifecycle_nodes[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_list_lifecycle_node_transitions[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_list_lifecycle_nodes[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_node_lifecycle[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_set_hidden_lifecycle_node_transition[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok
test_set_lifecycle_node_invalid_transition[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... FAIL
test_set_nonexistent_lifecycle_node_state[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI) ... ok

======================================================================
FAIL: test_get_hidden_lifecycle_node_state[rmw_cyclonedds_cpp] (test_cli.TestROS2LifecycleCLI)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/install/launch_testing/lib/python3.7/site-packages/launch_testing/markers.py", line 57, in _wrapper
    return func(self, *args, **kwargs)
  File "/Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/src/ros2/ros2cli/ros2lifecycle/test/test_cli.py", line 357, in test_get_hidden_lifecycle_node_state
    assert launch_testing.tools.expect_output(
AssertionError: assert False
 +  where False = <function expect_output at 0x1070fca70>(expected_lines=['unconfigured [1]'], text='1586517687.773897 [180]    3883402: using network interface en0 (udp/192.168.1.120) selected arbitrarily from: en0, en1\nunconfigured [1]\n', strict=True)
 +    where <function expect_output at 0x1070fca70> = <module 'launch_testing.tools' from '/Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/install/launch_testing/lib/python3.7/site-packages/launch_testing/tools/__init__.py'>.expect_output
 +      where <module 'launch_testing.tools' from '/Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/install/launch_testing/lib/python3.7/site-packages/launch_testing/tools/__init__.py'> = launch_testing.tools
 +    and   '1586517687.773897 [180]    3883402: using network interface en0 (udp/192.168.1.120) selected arbitrarily from: en0, en1\nunconfigured [1]\n' = <launch_testing.tools.process.ProcessProxy object at 0x1133152d0>.output
```

Is this a problem with `launch_testing`, with  `colcon`, or with CI's failure message parsing?

dirk-thomas, 2020-04-11:
> Is this a problem with launch_testing, with colcon, or with CI's failure message parsing?

`colcon` only shows `stderr` output by default. `stdout` output can be enabled which is the case in Jenkins.

Jenkins shows the data available in the `.xml` test result files.

What / how `launch_testing` reports any kind of feedback can be better answered by the people which designed / implemented that tool.

hidmic, 2020-04-16:
That's a consequence of the [`pytest` shim](https://github.com/ros2/launch/blob/a3db681a63a5908d38ca5fcac4bdd96f5e4fed7e/launch_testing/launch_testing/pytest/hooks.py#L58-L75) not propagating all available information properly. I agree with @rotu we have to iterate on this.