ros2/launch #411: ros2 node list launches 2 nodes of the same name and type

ddengster, 2020-04-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 1804
- Installation type:
  - Source
- Version or commit hash:
  - Head

#### Steps to reproduce issue
Follow https://answers.ros.org/question/344141/ros2-launch-creates-two-nodes-of-same-type/

#### Expected behavior
Only 1 node should be launched when doing `ros2 node list`

#### Actual behavior
2 nodes listed on `ros2 node list` with the same name.

#### Additional information
Removing the `node_name`(or in later versions `name`) argument from the launch file fixes it.
https://answers.ros.org/question/344141/ros2-launch-creates-two-nodes-of-same-type/

hidmic, 2020-05-07:
I've replied to those answer.ros.org questions. This seems connected to https://github.com/ros2/geometry2/issues/124 and it's not an issue with `geometry2` or `launch`, but with `robot_localization` and/or user code.

closed 2020-05-07 by hidmic

130s, 2023-04-15:
> I've replied to those answer.ros.org questions. This seems connected to [ros2/geometry2#124](https://github.com/ros2/geometry2/issues/124) and it's not an issue with `geometry2` or `launch`, but with `robot_localization` and/or user code.

I'm seeing the similar/same issue, in my program none of those packages you mentioned are involved. Ticketed https://github.com/ros2/launch/issues/704