ros2/launch #412: No error information provided when using frontend launch files

jdlangs, 2020-05-05:
## Bug report

This is not actually a bug but rather important missing functionality.

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: Eloquent from binaries, `launch` from source
- Version or commit hash: `cac43b195fb767128df1fdf494687973d538c792` (launch)

#### Steps to reproduce issue

Invalid launch xml file, e.g.:
```
<?xml version="1.0"?>
<launch>
  <node />
</launch>
```

#### Expected behavior

Some information provided about what line or tag caused a problem when loading the file. I would expect this to be provided as clean formatted output and not in a huge python traceback.

#### Actual behavior

A generic exception is thrown that the launch file is invalid and `ros2 launch` immediately terminates.

> [DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
> [DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
> [INFO] [launch]: Default logging verbosity is set to DEBUG
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fb1dc627438>'
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fb1dc627438>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7fb1d2119710>'
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fb1cdd67588>'
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fb1cdd67588>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7fb1d2119710>'
> [DEBUG] [launch]: Traceback (most recent call last):
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/launch_service.py", line 362, in run_async
>     await process_one_event_task
>   File "/usr/lib/python3.6/asyncio/coroutines.py", line 126, in send
>     return self.gen.send(value)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/launch_service.py", line 273, in _process_one_event
>     await self.__process_event(next_event)
>   File "/usr/lib/python3.6/asyncio/coroutines.py", line 110, in __next__
>     return self.gen.send(None)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/launch_service.py", line 293, in __process_event
>     visit_all_entities_and_collect_futures(entity, self.__context))
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
>     futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
>     sub_entities = entity.visit(context)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/action.py", line 108, in visit
>     return self.execute(context)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/actions/include_launch_description.py", line 125, in execute
>     launch_description = self.__launch_description_source.get_launch_description(context)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/launch_description_source.py", line 84, in get_launch_description
>     self._get_launch_description(self.__expanded_location)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
>     return get_launch_description_from_any_launch_file(location)
>   File "/home/jlangsfeld/Workspaces/ros2/build/launch/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
>     raise InvalidLaunchFileError(extension, likely_errors=exceptions)
> launch.invalid_launch_file_error.InvalidLaunchFileError: The launch file may have a syntax error, or its format is unknown
> 
> [ERROR] [launch]: Caught exception in launch (see debug for traceback): The launch file may have a syntax error, or its format is unknown
> [DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
> [DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb1c5be12e8>'
> [DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb1c5be12e8>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fb1c5bbc128>'
> [DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fb1c5be12e8>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fb1da7f6a90>'

#### Additional information

The same exception is also thrown when you do have a valid XML frontend file but don't happen to have the `launch_xml` package installed. It's a really terrible user experience that no information is provided at all about what the problem is.

hidmic, 2020-05-07:
This is a known TBD. XML and YAML launch frontend packages could do a better job reporting source errors.

corot, 2022-12-17:
Absolutely agree. As a partial palliative for the pain, I'm adding the schema on all my launch files:
```
<launch xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xsi:noNamespaceSchemaLocation="https://raw.githubusercontent.com/ros2/design/gh-pages/articles/specs/launch.0.1.1.xsd" version="0.1.1">
....
```
It helps a lot, but still falls short when you have several nested launch files.
Moreover, I don't think that's the last version, as it doesn't mention `push-ros-namespace`.

ijnek, 2025-06-16:
I'm sure many cases aren't covered, but the minimal reproducible example provided in the original post seems to now provide a useful error message:

```sh
$ ros2 launch test.launch.xml 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-06-16-00-06-15-007494-ijnek-ThinkPad-15007
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [xml]:
 - AttributeError: Attribute 'exec' of type '<class 'str'>' not found in Entity 'node'
 - SyntaxError: invalid syntax (test.launch.xml, line 1)
```

Interested to see other examples.