ros2/launch #418: Race between Shutdown and starting a new process

cottsay, 2020-05-08:
## Bug report

**Required Info:**

- Operating System:
  - Fedora 32
- Installation type:
  - from source
- Version or commit hash:
  - cac43b1
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
```python
import launch

def generate_launch_description():
    return launch.LaunchDescription([
        launch.actions.ExecuteProcess(cmd=[
            launch.substitutions.FindExecutable(name='sleep'), '10'
        ]),
        launch.actions.Shutdown(),
    ])
```

#### Expected behavior
Launch starts the sleep process and the immediately sends Ctrl+C to that process, then exits after a very short amount of time.

#### Actual behavior
Launch waits 5 seconds to SIGTERM the process and prints the following error:
<details>

```
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at launch/launch/launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "launch/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "launch/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "launch/launch/action.py", line 108, in visit
    return self.execute(context)
  File "launch/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "launch/launch/actions/execute_process.py", line 421, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
```

</details>

#### Additional information

This is a contrived example, but it demonstrates the race between a launch shutdown and a process being spawned. I imagine that you could get it to happen if you Ctrl-C at just the right time during a process launch.

Another way to trigger this is to create a launch test that runs the tests REALLY FAST (like just running `assert True` or something). Since even a simple test like that takes a little time, launch fails sometimes and succeeds others.

Maybe a ShutdownProcess event arrives before the process has even started? I'm not sure.

Blast545, 2020-05-12:
This problem occurs as well in Windows, https://github.com/ros2/demos/issues/438.

- Operating System:
  - Windows Server 2019 64bits
- Installation type:
  - from source
- Version or commit hash:
  - 7d813d4
- DDS implementation:
  - CycloneDDS, Fast-RTPS (static and dynamic)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
`colcon test --merge-install --packages-select composition`

<details>

```
Start  2: test_linktime_composition__rmw_cyclonedds_cpp
[13.797s] 
[13.797s] 2: Test command: C:\Python37\python.exe "-u" "C:/dev/ros2_ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/dev/ros2_ws/build/composition/test_results/composition/test_linktime_composition__rmw_cyclonedds_cpp.xunit.xml" "--package-name" "composition" "--output-file" "C:/dev/ros2_ws/build/composition/launch_test/CHANGEME.txt" "--env" "RMW_IMPLEMENTATION=rmw_cyclonedds_cpp" "--append-env" "AMENT_PREFIX_PATH=C:/dev/ros2_ws/build/composition/test_ament_index/Release" "PATH=C:/dev/ros2_ws/build/composition/Release" "--command" "C:/Python37/python.exe" "-m" "launch_testing.launch_test" "C:/dev/ros2_ws/build/composition/test_linktime_composition__rmw_cyclonedds_cpp_Release.py" "--junit-xml=C:/dev/ros2_ws/build/composition/test_results/composition/test_linktime_composition__rmw_cyclonedds_cpp.xunit.xml" "--package-name=composition"
[13.797s] 2: Test timeout computed to be: 60
[13.906s] 2: -- run_test.py: extra environment variables:
[13.906s] 2:  - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
[13.906s] 2: -- run_test.py: extra environment variables to append:
[13.906s] 2:  - AMENT_PREFIX_PATH+=C:/dev/ros2_ws/build/composition/test_ament_index/Release
[13.906s] 2:  - PATH+=C:/dev/ros2_ws/build/composition/Release
[13.906s] 2: -- run_test.py: invoking following command in 'C:\dev\ros2_ws\build\composition':
[13.906s] 2:  - C:/Python37/python.exe -m launch_testing.launch_test C:/dev/ros2_ws/build/composition/test_linktime_composition__rmw_cyclonedds_cpp_Release.py --junit-xml=C:/dev/ros2_ws/build/composition/test_results/composition/test_linktime_composition__rmw_cyclonedds_cpp.xunit.xml --package-name=composition
[14.328s] 2: [INFO] [launch]: All log files can be found below C:\Users\Administrator\.ros/log\2020-05-12-15-23-15-642868-EC2AMAZ-H300BQC-4124
[14.328s] 2: [INFO] [launch]: Default logging verbosity is set to INFO
[14.328s] 2: test_linktime_composition (composition.TestComposition)
[14.390s] 2: Test process' output against expectations. ... skipped ''
[14.390s] 2: 
[14.390s] 2: ----------------------------------------------------------------------
[14.390s] 2: Ran 1 test in 0.000s
[14.390s] 2: 
[14.390s] 2: OK (skipped=1)
[14.390s] 2: Task exception was never retrieved
[14.390s] 2: future: <Task finished coro=<LaunchService._process_one_event() done, defined at C:\dev\ros2_ws\install\Lib\site-packages\launch\launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
[14.390s] 2: Traceback (most recent call last):
[14.390s] 2:   File "C:\dev\ros2_ws\install\Lib\site-packages\launch\launch_service.py", line 273, in _process_one_event
[14.390s] 2:     await self.__process_event(next_event)
[14.390s] 2:   File "C:\dev\ros2_ws\install\Lib\site-packages\launch\launch_service.py", line 293, in __process_event
[14.390s] 2:     visit_all_entities_and_collect_futures(entity, self.__context))
[14.390s] 2:   File "C:\dev\ros2_ws\install\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
[14.390s] 2:     sub_entities = entity.visit(context)
[14.390s] 2:   File "C:\dev\ros2_ws\install\Lib\site-packages\launch\action.py", line 108, in visit
[14.390s] 2:     return self.execute(context)
[14.390s] 2:   File "C:\dev\ros2_ws\install\Lib\site-packages\launch\actions\opaque_function.py", line 75, in execute
[14.390s] 2:     return self.__function(context, *self.__args, **self.__kwargs)
[14.390s] 2:   File "C:\dev\ros2_ws\install\Lib\site-packages\launch\actions\execute_process.py", line 421, in __on_signal_process_event
[14.390s] 2:     raise RuntimeError('Signal event received before subprocess transport available.')
[14.390s] 2: RuntimeError: Signal event received before subprocess transport available.
[14.390s] 2: The link time registration of classes does not work on Windows
[14.390s] 2: [INFO] [test_linktime_composition-1]: process started with pid [584]
[14.468s] 2: [test_linktime_composition-1] [INFO] [1589296995.777844100] [linktime_composition]: Load library 
[19.343s] 2: [ERROR] [test_linktime_composition-1]: process[test_linktime_composition-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[19.343s] 2: [INFO] [test_linktime_composition-1]: sending signal 'SIGTERM' to process[test_linktime_composition-1]
[19.343s] 2: [ERROR] [test_linktime_composition-1]: process has died [pid 584, exit code 1, cmd 'C:/dev/ros2_ws/build/composition/Release/linktime_composition.exe'].
[19.343s] 2: 
[19.343s] 2: ----------------------------------------------------------------------
```
</details>

cottsay, 2020-05-12:
The composition test failure is a great example of this in the wild:
> Another way to trigger this is to create a launch test that runs the tests REALLY FAST...

...in the composition case, the test is skipped, which happens "really fast".

rotu, 2020-05-12:
Note that `LaunchDescription` doesn't even promise to execute its Entities in order, so this should probably be accompanied by a documentation update as well!

ivanpauno, 2020-06-19:
Fixed by https://github.com/ros2/launch/pull/427

closed 2020-06-19 by ivanpauno