ros2/launch #423: Files in example_processes missing `.py` extension

rotu, 2020-05-19:
The following Python files are missing the `.py` extension:

* launch_testing/example_processes/exit_code_proc
* launch_testing/example_processes/good_proc
* launch_testing/example_processes/terminating_proc

Since they are never invoked "naked" (e.g. always as `python3 /path/to/script`) there is no benefit to leaving off the `.py` extension to make them look like a native executable.

_Discovered in https://github.com/ament/ament_cmake/issues/213#issuecomment-631129869_

sloretz, 2020-05-29:
Thanks for reporting @rotu! Would you be willing to open a PR with a fix?

rotu, 2020-05-29:
@sloretz I'm trying not to open PRs for stuff like this that's off-task (hence this issue instead of a PR).

@pbaughman is listed as the package maintainer. Maybe he can help?

pbaughman, 2020-05-29:
Hi @rotu, I'm not currently working with the company who hired me to work on ROS2 testing.  It's OK for you to go ahead and make the change.

PR #573 (2022-01-03): Renamed three files from example_processes
The following Python files were missing the .py extension:

- launch_testing/example_processes/exit_code_proc
- launch_testing/example_processes/good_proc
- launch_testing/example_processes/terminating_proc

Fixes #423 

PS: My first GitHub PR, any suggestions/tips are welcome
closed 2022-01-03 by hidmic