ros2/launch #425: Outdated maintainer info

rotu, 2020-05-29:
The package.xml file lists the maintainer as Pete Baughmann, who is no longer the active maintainer of the package. 

REP 149 specifies that a package with no current maintainer should be marked as "orphaned" but I think it may be more appropriate to list the maintainer as Open Robotics.

https://www.ros.org/reps/rep-0149.html#maintainer-multiple-but-at-least-one

> Hi @rotu, I'm not currently working with the company who contracted with me to work on ROS2 testing.  It's OK for you to go ahead and make the change.

_Originally posted by @pbaughman in https://github.com/ros2/launch/issues/423#issuecomment-636146935_

https://github.com/ros2/launch/blob/dbbd4de641fa9f2116bd71600e2c8fe67f0c24e3/launch_testing/package.xml#L4-L11

pbaughman, 2020-05-29:
@rotu I have some availability now - if you guys are very busy and need some extra manpower on ROS2 testing, you can email me at peter.c.baughman@outlook.com to set up a contract.

rotu, 2020-05-30:
Thanks! I appreciate the willingness to help, though I think you misunderstand. I have a passing relation with this package, and I'm trying *not* to get too involved nor to fund significant changes at this time.

This issue is *just* about getting the package metadata correct and current.

sloretz, 2020-06-25:
@wjwwood friendly ping :)

wjwwood, 2020-06-25:
I'll take ownership for now: https://github.com/ros2/launch/pull/433

PR #433 (2020-06-25): claim ownership
fixes #425
commit db2a4217 (2020-06-25): claim ownership (#433)

fixes #425
closed 2020-06-25 by wjwwood