ros2/launch #441: Equivalent inspection options for ROS Launch files (`--dump-params`, `--nodes` and `--args`)

seanyen, 2020-07-16:
## Feature request

#### Feature description
In ROS1, the `roslaunch` provides a set of options (`--dump-params`, `--nodes` and `--args`) to inspect what's parameters and command lines that ROSLaunch eventually invoke. This is really useful for downstream tool to synthesize the ROS composition for debugging purpose. For example, [vscode-ros](https://github.com/ms-iot/vscode-ros/blob/master/doc/spec/debug-ros-nodes.md) extension is utilizing it to implement the ROSLaunch debugging feature, which is one of the mostly used functions.

It would be nice to see ROS 2 launch CLI can offer the similar experience, so the downstream tool can do the similar integration for ROS 2.

wjwwood, 2020-08-20:
Launch was designed to expose this information, but has yet to be utilized much yet. This example does do "introspection":

https://github.com/ros2/launch/blob/9e7be6cd07e2d5f6445a952990e9f9d0ac46b790/launch/examples/launch_counters.py#L127-L130

But again, it's not completely fleshed out as a feature. At that point those arguments can be implemented by categorizing the items visited during introspection.

I think this is not on our roadmap at the moment, but would be nice as an external contribution.

seanyen, 2020-08-20:
@wjwwood Thanks. And happy to see this type of improvement is welcome. My team is interested in making some contributions here. Maybe we can use ROS1 as a guide to define the options and the output format. Just a logistical question, where would you recommend us to begin with the proposal? (I am sure we probably want to have some discussions on that before we implement anything.)

wjwwood, 2020-08-20:
I think I'd like to see a pr adding to the package's documentation, e.g. a peer to this file:

https://github.com/ros2/launch/blob/master/launch/doc/source/architecture.rst

Or maybe something similar in the `ros2launch` or `launch_ros` packages.