ros2/launch #442: launch_ros_PID collisions in multi-host systems

dawonn-haval, 2020-07-19:
## Bug report

The launch_ros node uses the PID as a suffix to generate a 'unique' node name, but this is insufficient in multi host systems. I use docker swarm containers across a number of hosts with many name collisions:
[image]

**Required Info:**

- Operating System: U20.04
- Installation type: Binary (mostly...)
- Version or commit hash: Foxy
- DDS implementation: FAST_RTPS
- Client library (if applicable): N/A

#### Steps to reproduce issue
Launch ROS nodes on multiple hosts at system startup, or via a docker swarm.

#### Expected behavior
`ros2 node list` should have no warnings.

#### Actual behavior
`WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.`

#### Additional information
I wonder if we can simply use a random alpha-numeric suffix instead?

dawonn-haval, 2020-07-21:
https://github.com/ros2/geometry2/blob/ros2/tf2_ros/include/tf2_ros/static_transform_broadcaster.h#L61

Stumbled upon how the static_transform_publisher generates a random suffix. Should this be adopted here as well?

ivanpauno, 2020-07-21:
I would reuse the logic [here](https://github.com/ros2/launch/blob/9d42f115a3d59ede93260c46b9f107717c5d3a24/launch/launch/substitutions/anon_name.py#L73).

@dawonn-haval are you planning to open a PR addressing this?

dawonn, 2020-07-21:
Is on my list of things to eventually get to, although I'm not sure exactly when that will be...