ros2/launch #444: ROS launch respawn not available in dashing

muawijhe, 2020-07-31:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2020-08-03:
Are you talking about the feature in https://github.com/ros2/launch/pull/426 ?

Regardless, we don't typically backport features into already released distributions.  However, my understanding is that it is fairly straightforward to just include that functionality in your own launch scripts, so I'd suggest that as a workaround for now.

I'm going to go ahead and close this, but feel free to keep commenting.

closed 2020-08-03 by clalancette

muawijhe, 2020-08-03:
Dear Chris,

first of all, thanks for the reply :) I really appreciate your effort to explain why you will not do such a feature integration.

As I understand all the development efforts are migrating into the Foxy Fitzroy distribution and I agree, the show must go on.. but please try to follow my explanation before to reject these features request:

1. There are a lot of developers out there using Ubuntu (having a fantastic support for ROS1/ROS2);
2. Many products out there are still using ROS2 dashing as it is the most stable and long term ROS2 distribution for 18.04LTS (not all the products out there already migrated to the newest Ubuntu, and not all of them will do it either);
3. As the 18.04LTS is supported up to 2024 (for general support) and up to 2028 for security updates I am wondering why
the ROS2 distribution that should be supported up to May 2021 should be abandoned like you are doing...
4. As you said, this should be a simple backport for an **-essential-** functionality still missing in a distribution having still a year of support ahead;

My question for the ROS2 development group is the following:

Why are you asserting to support the distribution until a certain date if in practice as soon another distribution is released you abandon every and all features at once, and in addition you will not provide any further relevant support?

I don't wanna start a fight or a huge discussion, but please consider the following:

I am trying, as many other people/technology guys/CTOs/etc, to use ROS2 in robotics applications and products, but while I am relying completely on Ubuntu as operating system, because their distribution schedule is serious and support is really following the EOL of the products for real... 

I'm having problem with ROS2 as, once I have chosen a certain distribution for certain products (planning future releases and so on), considering a serious follow up until the EOL of the chosen release, what I am finding is a clearly less serious condition with a development community moving on very fast (which is a quality), but without looking behind to the people who are trying to rely on the release and the end of life date you actually suggested (which is really an issue for tech companies)....

So please, next time someone will try to suggest an "easy" feature backport on something that should actually be supported for another year, you could do a great service to the community, demonstrate that people can actually rely on the EOL dates and support of ROS2 community, instead of bouncing back the problem and suggest people to change ROS2 release (and consequently the OS version) instead.

Btw, I really appreciated the work you are doing and I believe ROS2 has a huge potential.

Hoping you will reconsider your decision for the sake of all the 18.04LTS guys out there ;)

V.

clalancette, 2020-08-03:
> Why are you asserting to support the distribution until a certain date if in practice as soon another distribution is released you abandon every and all features at once, and in addition you will not provide any further relevant support?

I don't think this is a fair assessment of the situation.  We definitely still support the older releases with bugfixes and new packages, up until the end-of-life date.  But we have limited resources to work on ROS 2, so it becomes a question of where to spend those resources.  We have generally decided to not spend them on backporting new features, and instead focus features on the newest releases.

> So please, next time someone will try to suggest an "easy" feature backport on something that should actually be supported for another year, you could do a great service to the community, [...]

I didn't suggest that you change OS or ROS distribution.  What I did say is that I believe this feature is relatively easy for you to work around in your own scripts, so I suggested that you do that instead.  For a bit more work on your side, you can continue to use Dashing on Ubuntu 18.04.  When you do upgrade to a newer release (whenever that may be), you can then remove these workarounds.

muawijhe, 2020-08-04:
Dear Chris,

Thanks again for the reply although we have clearly a very different
perspective about "new features". Respawn was a standard feature in ROS1
not an optional new cool feature. I believe it is correct to spend time
properly allocating resources on newest cool tasks, however here we are
talking about a "must" feature, not a "would like" feature.

That is why I am so surprised of your "laziness/friction" in "backporting"
such relevant (and still easy to backport) piece of code. To me it is
possible to adjust the roslaunch python module (although I am far to be a
good python programmer), but here I am talking for all the people out there
using ros dashing at least for another year.

I believe you (or any product manager of ROS2) could do a small effort, and
choose among all the new features the ones that are actually fundamental to
allow people out there to use ROS2 to build products.

That is what I am trying to pass you as message.

More fancy stuff are available in the newest release, cool ;) but the
needed ones should be also in the glorious dashing ;)

I believe that, given your replies, you could fix this issue in no time...
Perhaps I am wrong but you could make a good service for the dashing
community in no time ;)

Hope you will understand this perspective and while I am thankful for the
job you are doing for ROS2, I really hope you will fix this in the short
future.

I am dealing with an EU project of emergency robotics, where several ROS2
nodes are using to monitor the status of an active vulcano...

You could image, after working so many years with ROS1 without a single
issue (we have still products running on ROS1) my effort to initiate the
migration, convincing my tech guys about the benefits and so on....

And then I discover that respawn is not available in a still-supported
release... What a shame... Believe it or not, you will feel much safe on
this Vulcano knowing that is a sensor node crash will be restarted
automatically  ;)

Best regards

V.

jacobperron, 2020-08-05:
@muawijhe In my experience, if someone from the community proposes a backport PR for a feature that doesn't break ABI, there is a better chance it will be accepted.