ros2/launch #458: Eloquent Test failure caused by Deprecation warning?

sloretz, 2020-09-08:
## Bug report

**Required Info:**

- Operating System:
  - all
- Installation type:
  - nightlies
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - all
- Client library (if applicable):
  - all

#### Steps to reproduce issue

https://ci.ros2.org/job/ci_linux/12231/console
* Eloquent repos file
* Target distro Eloquent
* OS Bionic
* Build up to launch_testing
* test only launch_testing

#### Expected behavior

All tests would pass

#### Actual behavior

```
Starting >>> launch_testing
10:31:04 ============================= test session starts ==============================
10:31:04 platform linux -- Python 3.6.9, pytest-6.0.1, py-1.9.0, pluggy-0.13.1
10:31:04 cachedir: /home/jenkins-agent/workspace/ci_linux/ws/build/launch_testing/.pytest_cache
10:31:04 rootdir: /home/jenkins-agent/workspace/ci_linux/ws/src/ros2/launch/launch_testing, configfile: pytest.ini, testpaths: test
10:31:04 plugins: launch-testing-0.9.6, ament-copyright-0.8.1, ament-pep257-0.8.1, ament-flake8-0.8.1, ament-lint-0.8.1, colcon-core-0.6.0, cov-2.10.1, rerunfailures-9.1, repeat-0.8.0, mock-3.3.1
10:31:04 collecting ... 
10:31:04 collected 0 items / 1 error                                                    
10:31:04 
10:31:04 ==================================== ERRORS ====================================
10:31:04 ________________________ ERROR collecting test session _________________________
10:31:04 ../../../../../venv/lib/python3.6/site-packages/pluggy/hooks.py:286: in __call__
10:31:05     return self._hookexec(self, self.get_hookimpls(), kwargs)
10:31:05 ../../../../../venv/lib/python3.6/site-packages/pluggy/manager.py:93: in _hookexec
10:31:05     return self._inner_hookexec(hook, methods, kwargs)
10:31:05 ../../../../../venv/lib/python3.6/site-packages/pluggy/manager.py:87: in <lambda>
10:31:05     firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
10:31:05 ../../../../../venv/lib/python3.6/site-packages/_pytest/python.py:196: in pytest_collect_file
10:31:05     path=path, parent=parent
10:31:05 ../../../../../venv/lib/python3.6/site-packages/pluggy/hooks.py:286: in __call__
10:31:05     return self._hookexec(self, self.get_hookimpls(), kwargs)
10:31:05 ../../../../../venv/lib/python3.6/site-packages/pluggy/manager.py:93: in _hookexec
10:31:05     return self._inner_hookexec(hook, methods, kwargs)
10:31:05 ../../../../../venv/lib/python3.6/site-packages/pluggy/manager.py:87: in <lambda>
10:31:05     firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
10:31:05 launch_testing/pytest/hooks.py:114: in pytest_pycollect_makemodule
10:31:05     return pytest.Module(path, parent)
10:31:05 ../../../../../venv/lib/python3.6/site-packages/_pytest/nodes.py:95: in __call__
10:31:05     warnings.warn(NODE_USE_FROM_PARENT.format(name=self.__name__), stacklevel=2)
10:31:05 E   pytest.PytestDeprecationWarning: Direct construction of Module has been deprecated, please use Module.from_parent.
10:31:05 E   See https://docs.pytest.org/en/stable/deprecations.html#node-construction-changed-to-node-from-parent for more details.
10:31:05 - generated xml file: /home/jenkins-agent/workspace/ci_linux/ws/build/launch_testing/pytest.xml -
10:31:05 =========================== short test summary info ============================
10:31:05 ERROR  - pytest.PytestDeprecationWarning: Direct construction of Module has b...
10:31:05 !!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
10:31:05 =============================== 1 error in 0.15s ===============================
10:31:05 Failed   <<< launch_testing [0.96s, exited with code 2]
10:31:05 
10:31:05 Summary: 0 packages finished [1.81s]
10:31:05   1 package failed: launch_testing
10:31:05 
10:31:05 II> colcon test returned: '2'
10:31:05 # END SUBSECTION
```

#### Additional information
Possibly would be fixed by backporting https://github.com/ros2/launch/pull/391 to ROS Eloquent

sloretz, 2020-09-08:
This appears to be the cause of at least the ROS Eloquent nightly release job failures

hidmic, 2020-12-16:
@sloretz can this ticket be closed?

sloretz, 2020-12-16:
I think so. Closing.

PR #459 (2020-09-08): [Eloquent backport] stop using constructors deprecated in pytest 5.4 (#391) Switch to from_parent (partial #421) avoid deprecation warning, use from_parent (#402)
Cherry pick of #391 - checking if this fixes #458

Now also includes a cherry-pick of one commit from #421: f1c02f30e2bca3e461fb574df8e0694ee8d03f23

Now also includes a cherry-pick of #402
closed 2020-12-16 by sloretz