ros2/launch #466: [launch_testing]: Parameterized gtests

sjahr, 2020-10-23:
## Feature request
Enable passing parameters from a launch file into gtests launched with launch_test.
#### Feature description
I could not figure out how to pass parameters (e.g. loaded from a yaml file) from a launch file into a launch_test launched gtest. Apparently, this seems to be currently not possible (please correct me if I am mistaken), and [others encountered that problem too](https://answers.ros.org/question/360844/ros2-launch_test-gtest-parameters-argument/).
But I think that could be a great addition to the launch_testing framework. 

Thanks in advance!

hidmic, 2020-10-23:
Thanks for reporting @sjahr ! 

I'm going to go ahead and assume you've been using `GTest` actions and the tests under the `test_launch_testing` package as a basis. Don't. `launch_testing` was refactored several times, and right now those actions will not really get their return codes checked. I'll see we do something about them (at the very least, avoid further confusion). CC @wjwwood @ivanpauno for awareness.

---

For your specific use case, what you can do is to start your `gtest` node executable as a `Node`, and then monitor its return code.
 
```python
import unittest

import launch
import launch_ros
import launch_testing

import pytest

@pytest.mark.rostest
def generate_test_description():
    test_node = launch_ros.actions.Node(
        executable=path_to_gtest_node_executable,
        parameters=[path_to_params],
    )

    return launch.LaunchDescription([
        # other fixture actions
        test_node,
        launch_testing.actions.ReadyToTest(),
    ])

@launch_testing.post_shutdown_test()
class TestOutcome(unittest.TestCase):

    def test_exit_codes(self, proc_info):
        launch_testing.asserts.assertExitCodes(proc_info) 
```

Which is definitely not as straightforward as `rostest`, but much more powerful.

sjahr, 2020-10-26:
I will try this out. Thank you for your quick and detailed answer!

closed 2020-10-26 by sjahr

tylerjw, 2021-01-22:
@hidmic how would you go about setting a timeout for the test using this method?

hidmic, 2021-01-22:
> how would you go about setting a timeout for the test using this method?

I *think* It'd be much better to have this exchange in [answers.ros.org](http://answers.ros.org/) rather than in an old, closed ticket. WDYT?

tylerjw, 2021-01-22:
@hidmic Good idea, here: https://answers.ros.org/question/370082/how-to-set-timeout-a-for-a-gtest-in-ros2/