ros2/launch #471: The substitution exception hides the error message

TakaHoribe, 2020-11-11:
## Bug report

**Required Info:**

- Operating System: Ubuntu20.04 (Docker)
- Installation type: Binaries
- Version or commit hash: `0.10.2-1focal.20200921.185945`
- DDS implementation:
- Client library (if applicable):

#### Steps to reproduce issue

When I run the `xacro` command in launch file like below,

```
<param name="robot_description" value="$(command 'xacro $(var model) />
```

I've got this error message.

```
exception=SubstitutionFailure('executed command failed. Command: xacro /home/horibe/AutowareArchitectureProposal/install/vehicle_launch/share/vehicle_launch/urdf/vehicle.xacro ')>
```

The reason for this error was just the loaded xacro file is invalid, but the problem is the internal error message is hidden by the `SubstitutionFailure` exception. No other error information was output except for the above. So I couldn't know the reason why the command was failed.

On the other hand, when I run the `xacro` command directly in terminal, I've got this message.

```
horibe@ade:~/AutowareArchitectureProposal (ros2 *$% u=)$ xacro /home/horibe/AutowareArchitectureProposal/install/vehicle_launch/share/vehicle_launch/urdf/vehicle.xacro 
xacro:arg: missing attribute 'default'
when processing file: /home/horibe/AutowareArchitectureProposal/install/vehicle_launch/share/vehicle_launch/urdf/vehicle.xacro
```

In this case, the error will be properly printed as `xacro:arg: missing attribute 'default'`.

I dug the code and found that when the `result.returncode` is not `0`, it returns an exception here
https://github.com/ros2/launch/blob/master/launch/launch/substitutions/command.py#L115-L116

But the `result.stderr` is printed after the `result.returncode` check here.
https://github.com/ros2/launch/blob/master/launch/launch/substitutions/command.py#L117-L124

So the error message will not be printed.

When I switch the order of the `if result.returncode != 0:` (L115-L116)  and `if result.stderr:` (L117-L124), I could get the proper error message like below.

```
...
/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=SubstitutionFailure("executed command showed stderr output. Command: xacro 
...
...
launch.substitutions.substitution_failure.SubstitutionFailure: executed command showed stderr output. Command: xacro /home/horibe/AutowareArchitectureProposal/install/vehicle_launch/share/vehicle_launch/urdf/vehicle.xacro 
Captured stderr output:
error: xacro:arg: missing attribute 'default'
when processing file: /home/horibe/AutowareArchitectureProposal/install/vehicle_launch/share/vehicle_launch/urdf/vehicle.xacro
```

I don't know this is the right solution, so I just report as an issue here.

#### Expected behavior
Outputs the proper error message.

#### Actual behavior
The error message is hidden by the exception.

hidmic, 2020-11-12:
@TakaHoribe thanks for digging this bug up! Ensuring captured `stderr` goes into the [`SubstitutionFailure` exception](https://github.com/ros2/launch/blob/79515f5ef4d3d1739d9bfb8002c833cfbb98324e/launch/launch/substitutions/command.py#L116) sounds best. Would you be willing to contributing such a fix?

TakaHoribe, 2020-11-18:
@hidmic Sure! I'll try this when I'm available (maybe weekend). But it would be also great if someone who is familiar with this code will fix this issue.

hidmic, 2020-12-01:
Addressed by #474.

closed 2020-12-01 by hidmic