ros2/launch #481: Launch argument with restricted choices

v-lopez, 2021-01-20:
## Feature request

#### Feature description
Launch argument that accepts only a few predetermined arguments.
For instance argument `laser_model` would only accept one of "sick-551, sick-561 or sick-571". 

#### Implementation considerations
I actually have already implemented this for our internal use, and I was wondering whether it'd be interesting to have here, before going through the hassle of adding tests and polishing enough to be on this repo.

The current implementation is below, probably can be improved a lot.
```python
class DeclareLaunchArgumentChoice(DeclareLaunchArgument):
    """
    Action that declares a launch argument with a limited range of choices
    """                                                                                                                                                                                                                                                                            
    def __init__(                                                                                                                                                                                                                                                                  
        self,                                                                                                                                                                                                                                                                      
        name: Text,                                                                                                                                                                                                                                                                
        *,                                                                                                                                                                                                                                                                         
        default_value: Optional[SomeSubstitutionsType] = None,                                                                                                                                                                                                                     
        description: Text = 'no description given',                                                                                                                                                                                                                                
        arg_choices: List[Text],                                                                                                                                                                                                                                                   
        **kwargs                                                                                                                                                                                                                                                                   
    ) -> None:                                                                                                                                                                                                                                                                     
        """Create a DeclareLaunchArgumentChoice action."""                                                                                                                                                                                                                         
        expanded_description = description + \                                                                                                                                                                                                                                     
            " Valid choices are: " + str(arg_choices)                                                                                                                                                                                                                              
        super().__init__(name, default_value=default_value,                                                                                                                                                                                                                        
                         description=expanded_description, **kwargs)                                                                                                                                                                                                               
        self.__arg_choices = arg_choices                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                   
    def execute(self, context: LaunchContext):                                                                                                                                                                                                                                     
        super().execute(context)                                                                                                                                                                                                                                                   
        value = context.launch_configurations[self.name]                                                                                                                                                                                                                           
        if value not in self.__arg_choices:                                                                                                                                                                                                                                        
            error_msg = "Argument '{}' provided value '{}' is not valid. Valid options are: {}".format(                                                                                                                                                                            
                self.name, value, self.__arg_choices)                                                                                                                                                                                                                              
            # Argument not already set and no default value given, error.                                                                                                                                                                                                          
            self._DeclareLaunchArgument__logger.error(error_msg)                                                                                                                                                                                                                   
            raise RuntimeError(error_msg)                                                                                                                                                                                                                                          
```

hidmic, 2021-01-20:
@v-lopez that'd be a nice feature. It could be added to the current `DeclareLaunchArgument` action, as an optional. Would you be willing to contribute a PR?

v-lopez, 2021-01-20:
Sure, probably not before this Friday though.

closed 2021-02-09 by hidmic