ros2/launch #484: ShutdownProcess event does not work from TimerAction

jacobperron, 2021-01-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- Version or commit hash:
  - 0.15.0

#### Steps to reproduce issue

launch file (`repro.launch.py`):
```
import launch
from launch.actions import ExecuteProcess
from launch.actions import EmitEvent
from launch.actions import TimerAction
from launch.events.process import ShutdownProcess
from launch.events import matches_action

def generate_launch_description():
    execute_proc = ExecuteProcess(
        cmd=['./test.sh'],
        name='my_test',
        output='screen',
    )
    kill_proc_event = EmitEvent(
        event=ShutdownProcess(
            process_matcher=matches_action(execute_proc)
        )
    )
    return launch.LaunchDescription([
        execute_proc,
        TimerAction(period=0.5, actions=[kill_proc_event]),
        # replacing the timer above with the following line works as expected (but without the time delay)
        # kill_proc_event
    ])
```

test process `test.sh`:

```
#!/usr/bin/env bash

echo "Test!"
sleep 1
echo "One"
sleep 1
echo "Two"
sleep 1
echo "Three"
```

launch command:

```
ros2 launch repro.launch.py
```

#### Expected behavior

After the timer expires, the ExecuteProcess action is shutdown.

#### Actual behavior

We see a SIGINT being sent to the process, but it is not shutdown.

#### Additional information

If we remove `TimerAction`, and emit the shutdown signal directly (see commented line in example), then the launch script behaves as expected: the test process is immediately shutdown.

jacobperron, 2021-01-26:
If I increase the delays in `test.sh`, eventually launch gives up on the SIGINT and sends a SIGTERM. This successfully kills the process:

```
[ERROR] [my_test-1]: process[my_test-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
```

jacobperron, 2021-01-26:
@hidmic I know you where dealing with signal handlers and launch recently, any ideas? Note, this doesn't look like a regression, just something that never worked.

hidmic, 2021-01-27:
I may be wrong, but this doesn't seem like a problem in `launch`.  `launch` sends signals to your test process, but your test process does not forward signals to its child processes. Ctrl+C on a terminal with job control is not the same as SIGINT'ing the process. This [blog](https://www.vidarholen.net/contents/blog/?p=34) talks about it briefly.

jacobperron, 2021-01-28:
You're right, launch is working as expected. I was not understanding how the terminal was interacting differently than sending a SIGINT. TIL. Thanks!

Handling the interrupt with `trap` or swapping in a Python script works as expected.

closed 2021-01-28 by jacobperron