ros2/launch #488: Setting log Debug level works only in code

ralwing, 2021-02-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04...) 
- Installation type:
  - official foxy docker image
- Version or commit hash:
- DDS implementation: rmw_fastrtps_cpp
- Client library (if applicable):

#### Steps to reproduce issue
create a ComposableNode
```
import launch
from launch_ros.actions import ComposableNodeContainer, Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    container = ComposableNodeContainer(
        name="container",
        namespace="",
        package="rclcpp_components",
        executable="component_container",
        composable_node_descriptions=[

            ComposableNode(
                package="localization",
                plugin="LocalizationNode",
                name="localization"
            )
        ],
        arguments=['--ros-args', '--log-level', 'DEBUG'] 
    )

    return launch.LaunchDescription([container])

```

#### Expected behavior
The log level is set to debug
#### Actual behavior
The log level doesn't change
#### Additional information
There are some topics related to this on answers.ros.org:
https://answers.ros.org/question/311471/selecting-log-level-in-ros2-launch-file/
https://answers.ros.org/question/370288/ros2-how-do-you-set-verbosity-of-a-composable-node/#370389

https://github.com/ros2/launch/issues/383

Setting log level to at least info works fine, setting log level in source code also works:
```
rcutils_ret_t rcutils_logging_set_logger_level (const char *name, int level);
```

----
## Feature request

Allow setting Debug severity log.

fujitatomoya, 2021-02-25:
> Setting log level to at least info works fine

i do not think so, this is default level so it just seems to work.
we can check that with `arguments=['--ros-args', '--log-level', 'WARN']`, this should not print any `[INFO]` messages.

as far as i can see from the source code, ComposableNodeContainer(Node).arguments are assigned to be user-defined arguments, not ros specific command line flags.

https://github.com/ros2/launch_ros/blob/d1c497878274a4ea6a1c1865fe74e395b961ba1f/launch_ros/launch_ros/actions/node.py#L146-L149

hidmic, 2021-03-09:
This may be due to components having their own configuration separate from that of the container (see [here](https://github.com/ros2/rclcpp/blob/6cc89caa639bc3f5ef543cf7440dafec588dab05/rclcpp_components/src/component_manager.cpp#L150-L171)). https://github.com/ros2/rclcpp/issues/978 is tracking that limitation, though I haven't got around to do what I said I was going to do then.