ros2/launch #492: "Command" substitution do not exist in the doc file

KarasevPavel, 2021-03-23:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

hidmic, 2021-03-26:
@KarasevPavel would you mind to be a bit more specific? Thanks!

KarasevPavel, 2021-03-27:
@hidmic  in this [document](https://github.com/ros2/launch/blob/master/launch/doc/source/architecture.rst) there is no information about `launch.substitutions.Command()` substitution, which was emplemented

hidmic, 2021-03-29:
@KarasevPavel that document describes the architecture of `launch`, it is not an exhaustive API reference. Unfortunately, there isn't any API reference hosted online just yet. Calling Sphinx on the package may be your best shot right now.

I'll close this ticket now. Feel free to open another one if you're unable to generate documentation locally. Moreover, contributions are most welcome if you have a fix to share.

closed 2021-03-29 by hidmic

kurshakuz, 2021-11-02:
@hidmic is there any way to build the API reference locally at the least? Building in the `doc` folder only shows the Architecture page.

I am also quite confused about the `Command` substitution and want to learn more about it. I tried to passing it a simple cmd commands in the ExecuteProcess and it returning me errors. like:
```
launch.substitutions.substitution_failure.SubstitutionFailure: executed command showed stderr output. Command: ros2 param set /sim background_r 120
```

However, I have some working examples of using it for obtaining a URDF string using xacro command, but I am curios what are other possibilities of the Command substitution.

hidmic, 2021-11-02:
> @hidmic is there any way to build the API reference locally at the least? Building in the doc folder only shows the Architecture page.

Hmm, `autodoc` output should be showing. I'll take a look.

>  I tried to passing it a simple cmd commands in the ExecuteProcess and it returning me errors.

That's likely due to the default behavior on stderr. You can [change](https://github.com/ros2/launch/blob/edcca09e4f80278d1a76e9c30eefbeeece5b33fd/launch/launch/substitutions/command.py#L53) that.