ros2/launch #496: Allow less verbose "IncludeLaunchDescription"

caioaamaral, 2021-03-30:
## Feature request
One could optionally pass the path to the launch file being included. In such case,  [`IncludeLaunchDescription`](https://github.com/ros2/launch/blob/master/launch/launch/actions/include_launch_description.py) would receive it as an [`AnyLaunchDescriptionSource`](https://github.com/ros2/launch/blob/bfe5ec2101ce1a5e46e5fadd95af546925350621/launch/launch/launch_description_sources/any_launch_description_source.py).
#### Feature description
I'm expecting this usage:
```py
from launch.actions import IncludeLaunchDescription
from launch.substitutions import PathJoinSubstitution

def generate_launch_description():
    return LaunchDescription(
        IncludeLaunchDescription('path/to/my/launch.[py|xml|yaml]')
    )

```
This would require a little change at  [`IncludeLaunchDescription`](https://github.com/ros2/launch/blob/master/launch/launch/actions/include_launch_description.py), making `launch_description_source` accept both `LaunchDescriptionSource` and `SomeSubstitutionsType` (launch path)
```py
from ..utilities import normalize_to_launch_description_source

@expose_action('include')
class IncludeLaunchDescription(Action):

    def __init__(
        self,
        launch_description_source: Union[LaunchDescriptionSource, SomeSubstitutionsType],
        *,
        launch_arguments: Optional[
            Iterable[Tuple[SomeSubstitutionsType, SomeSubstitutionsType]]
        ] = None,
        **kwargs
    ) -> None:
        """Create an IncludeLaunchDescription action."""
        super().__init__(**kwargs)
        self.__launch_description_source = \
            normalize_to_launch_description_source(launch_description_source)
        ...
```
A helper method could be used for the dynamic dispatch:
```py
from ..launch_description_sources import AnyLaunchDescriptionSource

def normalize_to_launch_description_source(source: Union[LaunchDescriptionSource, SomeSubstitutionsType]) -> LaunchDescriptionSource:

    if isinstance(source, LaunchDescriptionSource):
        return source
    if isinstance(source, SomeSubstitutionsType):
        return AnyLaunchDescriptionSource(launch_file_path=source)
    raise TypeError(
        "Failed to normalize given item of type '{}', when only "
        "''launch.LaunchDescriptionSource' or 'launch.AnyLaunchDescriptionSource' were expected.".format(type(x)))
```
#### Implementation considerations
Would like to know if people are also bothered with always having to expend extra lines for launch inclusions (I mean, always calling `IncludeLaunchDescription` followed by a `PythonLaunchDescriptionSource`). If this feature is desired, I can submit a pr.

hidmic, 2021-04-05:
Looks like nice feature to have to me.

brennand, 2021-07-05:
I wrote something similar:

```
    ur_launch = IncludeLaunchDescription(
        get_launch_file(ThisLaunchFileDir(), "/picknik_ur5e.launch.py"),
        launch_arguments={"robot_ip": robot_config["ip"]}
    )
```

`get_launch_file` just returns a `PythonLaunchDescriptionSource`

Khush-dev, 2021-12-25:
It looks like this issue has been addressed in #544 and implemented too.
If there's something that I missed, let me know

hidmic, 2022-01-03:
Indeed @Khush-dev. @caioaamaral feel free to re-open if this isn't enough for your use case.

closed 2022-01-03 by hidmic