ros2/launch #499: Launch Loop Support? [Feature Request]

shonigmann, 2021-04-02:
## Feature request
Not knowing too much about what's going on under the hood, would it be possible to support loops based on launch configuration values? 
 
#### Feature description

Similar to how launch conditions can be used to decide whether or not to spawn a node, it would be helpful to be able to spawn N nodes, based on a user or CLI set launch configuration. In particular, this could be useful for simulating swarms of robots of different sizes, on the fly.

#### Implementation considerations

I am aware of two possible work-arounds; 
1) creating a helper function that calls the single-instance launch command N times with an additional namespace variable passed in 
2) Adding N_max if-conditional statements in the launch file and comparing them to a passed in launch configuration. 

The former feels cleaner, but it would be nice to have a solution that exists entirely in the launch file.

hidmic, 2021-04-05:
Hmm, I'm not entirely convinced we want to walk towards Turing complete launch descriptions. I do understand your use case, but I wonder if it should be addressed in configuration time instead i.e. by the `get_launch_description()` function for Python launch files, by a preprocessor for XML launch files, and so on.

shonigmann, 2021-04-05:
I'll happily defer to those with a better understanding of the potential drawbacks than myself. For me, its a small _nice-to-have_, but by no means essential to what I'm doing. So if it doesn't make sense to implement I am happy to close the issue.

hidmic, 2021-04-22:
Circling back. @shonigmann if you'd like to move forward with this feature, as per our [development workflow](https://docs.ros.org/en/foxy/Contributing/Developer-Guide.html#developer-workflow), we'll need a design plus compelling use cases for the feature. As mentioned above, there are concerns around the complexity that built-in loop support in launch descriptions would introduce, so we have to sort those out.

Alternatively, and as suggested offline by @wjwwood (good tip!), you could use an [`OpaqueFunction`](https://github.com/ros2/launch/blob/3a5c9df2e384271637ac029f48f020b7916498d6/launch/launch/actions/opaque_function.py#L32) to run arbitrary Python code within `launch`, which can generate and return newly created actions in run-time. Something along the lines of:

```py
def include_swarm_launch_descriptions(context):
    return [
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([ThisLaunchFileDir(), '/robot.launch.py']),
            launch_arguments={'robot_id': n}.items(),
        ) for n in range(int(context.launch_configurations['num_robots']))
    ]

def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument('num_robots'),
        # ...
        OpaqueFunction(include_swarm_launch_descriptions),
        # ...
    ])
``` 

might do.

shonigmann, 2021-04-22:
@wjwwood - I appreciate the added feedback. For my own use cases, I would use this functionality pretty sparingly, so I am happy enough to use OpaqueFunction as a work around with some verbose comments to describe what's happening. I could quickly poll the community over on ROS Discourse though to get a better sense of whether other users would be interested in such a feature for their own applications and to potentially (de/)motivate the new launch action.