ros2/launch #509: LaunchConfigurationNotEquals() always returns `True`

HMellor, 2021-05-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ros-foxy-launch: 0.10.5-1focal.20210423.000156
- DDS implementation:
  - (Don't know how to check but don't think it's relevant here?)
- Client library (if applicable):
  - launch

#### Steps to reproduce issue
I am trying to include an external launch file in a new launch file, but only if an argument in the new launch file is set. I have tried to do this using the `condition` kwarg where the specified condition evaluates to `False` if the value of `arg` is the default `""`. However, the condition always evaluates to `True`.

```python
package_dir = ament_index_python.packages.get_package_share_directory("package")

declare_ncom = launch.actions.DeclareLaunchArgument("arg", default_value="")
arg = launch.substitutions.LaunchConfiguration("arg", default="")

launch_description = launch.launch_description_sources.PythonLaunchDescriptionSource(f"{package_dir}/launch/launch.py")
package_launch = launch.actions.IncludeLaunchDescription(
    condition=launch.conditions.LaunchConfigurationNotEquals(arg, ""), 
    launch_description_source=launch_description,
    launch_arguments={"arg": arg}.items(),
)
```

#### Expected behavior
If the `arg` has not been set, then `package_launch`'s condition should evaluate to `False` and the `launch.py` should not run from the new launch file.

#### Actual behavior
`launch.py` always runs.

hidmic, 2021-05-31:
@HMellor [`LaunchConfigurationNotEquals` takes the launch configuration name](https://github.com/ros2/launch/blob/b59bb6b88b40eaa82870542d93bdfb1a033d05c3/launch/launch/conditions/launch_configuration_not_equals.py#L41), not the object itself. 

Having said that, I'm open to supporting that OR improving documentation to make it clearer. Would you be willing to do the latter?

HMellor, 2021-05-31:
Ah ok, the type hinting for the argument was [`SomeSubstitutionsType`](https://github.com/ros2/launch/blob/b59bb6b88b40eaa82870542d93bdfb1a033d05c3/launch/launch/some_substitutions_type.py#L24), which is non-specific and open to being misinterpreted (such as I have done here). If it's just the docstring I'd be happy to make a pull request to clarify the argument. Have I understood you correctly @hidmic?

hidmic, 2021-05-31:
> If it's just the docstring I'd be happy to make a pull request to clarify the argument. Have I understood you correctly @hidmic?

Yes, exactly.

PR #510 (2021-06-14): Improve (Not)Equals condition type hinting
Fixes #509
closed 2021-06-14 by hidmic