ros2/launch #513: Documentation not published to docs.ros2.org/galactic/api

russkel, 2021-06-21:
The various launch package documentation is not published. Given the complexity of the launch system perhaps it would be good to get them up there.

ar13pit, 2021-11-18:
Is any documentation of this package published? I'm unable to find it and have to go through the rst files in the repository.

clalancette, 2021-11-18:
Not currently, no.  This is on the roadmap to get in for the next release.

andrewjong, 2024-10-17:
Bump, any updates to adding documentation on launch_testing to https://docs.ros.org/? How can we help?

clalancette, 2024-10-17:
If you'd like to contribute a new page about `launch_testing` to the testing page, that would be wonderful.  I'd be happy to review it.

EmmanuelMess, 2025-02-07:
Are there any news on adding the API to the docs?