ros2/launch #519: Feature request: make timeout from launch_testing's process_start configurable

MarcoLm993, 2021-07-13:
## Feature request

#### Feature description
Right now launch_tests wait only 15 seconds from the time the last process starts and the ReadyToTest action gets called.
This value is hard-coded in launch_testing and can be found [here](https://github.com/ros2/launch/blob/master/launch_testing/launch_testing/test_runner.py#L184).
Even though most times this timeout is large enough, there might be some larger systems that take longer to get ready before the actual test is ready to be executed. In those cases the whole test will fail saying `Timed out waiting for processes to start up`.

It would be nice to have the possibility to configure the timeout from the command line, in order to set a larger one in case it is necessary.

marekmlk, 2024-01-30:
Already implemented in #625 for rolling.

MarcoLm993, 2024-01-30:
Then I guess this issue can be closed :)

closed 2024-01-30 by MarcoLm993