ros2/launch #535: No output from roslaunch python

chngdickson, 2021-10-08:
## No output from roslaunch python

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 
```

```

#### roslaunch prints outputs from nodes

#### no logs are printed

#### 
[image]
Running the node using ros2 run works fine
[image]

hidmic, 2021-10-08:
Thank you for your bug report @chngdickson. We'll take a look.

ijnek, 2025-06-15:
Was just going through the issues - this issue sounds like a duplicate of https://github.com/ros2/launch/issues/188