ros2/launch #551: Logs are not saved in the same folder

DanyResasco, 2021-11-02:
- Operating System:
  Ubuntu 20
- Installation type:
  from source 

#### Expected behavior
As  ros1 all the logs should be saved as

logs/day-months-time_id/ [list of files]

#### Actual behavior
The logs are being saved in the logs folder but not in the same subfolder and only cpp nodes have the name of the file.

[image]

#### Additional information
for the python logs we use the library logging and for cpp the RCLCPP_*_STREAM

Do you know how to solve this problem?

hidmic, 2021-11-02:
Thing is that while `launch` logs go through the `logging` module, `rclcpp` and `rclpy` logs go through `rcl` logging (typically `spdlog`). Logging directories indeed do not match, as seen [here](https://github.com/ros2/launch/blob/edcca09e4f80278d1a76e9c30eefbeeece5b33fd/launch/launch/logging/__init__.py#L123) and [here](https://github.com/ros2/rcl_logging/blob/8d5ad0f9f32c1c3399122e28cf3e70f89ff0bc2f/rcl_logging_interface/src/logging_dir.c#L25). 

To get the functionality you describe, `launch` needs to force a logging base path to the processes it manages and there's no way to do that externally. I'm onboard with the idea though. @ros2/team thoughts?

wjwwood, 2021-11-29:
> To get the functionality you describe, launch needs to force a logging base path to the processes it manages and there's no way to do that externally today. I'm onboard with the idea though. @ros2/team thoughts?

+1 from me, though we'd need to consider how this would impact existing users and/or have an opt-in or opt-out setting for this behavior. At the very least we'd want to be loud about the change since some people may be relying on the layout of log messages.

daisukes, 2021-11-30:
Recently, I just got the same question.
I think we can use the `ROS_LOG_DIR` env variable to let nodes change their log directory inside the launch description.
```
from launch.logging import launch_config

def generate_launch_description():
    return LaunchDescription([
        SetEnvironmentVariable('ROS_LOG_DIR', launch_config.log_dir),
        ...
    ])
```

I'm also interested in the point @DanyResasco mentioned.
> only cpp nodes have the name of the file.

Each log's filename is defined by `rcutils_get_executable_name`, which makes all python-based node's log filename to `python3_<pid>_<time>.log`.
I'm using a c++ executable for multiple nodes in different namespaces, but the log file names are both in the same name, which is confusing.

https://github.com/ros2/rcutils/blob/5d3cecc8af601c10561f7c17d76eab740bb540ce/src/process.c#L56
https://github.com/ros2/rcl_logging/blob/8d5ad0f9f32c1c3399122e28cf3e70f89ff0bc2f/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp#L108

It would be good if it uses the name based on `__ns` and `__node` ros arguments.
If you run `ps -ef` you may see like
```
/<install_path>/<ros2_pkg>/lib/<ros2_pkg>/<executable name> --ros-args -r __node:=<node_name> -r __ns:=<namespace> ...
/usr/bin/python3 /<install_path>/<ros2_pkg>/lib/<ros2_pkg>/<python_module_name> --ros-args -r __node:=<node_name> -r __ns:=<namespace> ...
```

wjwwood, 2022-09-26:
This is partially addressed with this pr: https://github.com/ros2/launch_ros/pull/325