ros2/launch #564: Multiple SetEnvrionmentVariable override each other when launching multiple nodes

fas2037, 2021-12-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version:
  - Galactic
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue

Generate multiple nodes and call SetEnvironmentVariable in the LaunchDescription:
```
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import SetEnvironmentVariable

def generate_launch_description():
    return LaunchDescription([
        SetEnvironmentVariable(name='TEST', value='0'),
        Node(
            package="test_0",
            executable="test_0",
            name="test_0_node",
            output="screen",
        )
    ])
```
```
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import SetEnvironmentVariable

def generate_launch_description():
    return LaunchDescription([
        SetEnvironmentVariable(name='TEST', value='1),
        Node(
            package="test_1",
            executable="test_1",
            name="test_1_node",
            output="screen",
        )
    ])
```
Launch both nodes:
```
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():

    nodes = ["test_0", "test_1"]
    includes = []

    for n in nodes:
        includes.append(launch.actions.IncludeLaunchDescription(launch.launch_description_sources.PythonLaunchDescriptionSource(get_package_share_directory(n) + '/launch/' + n + '_launch.py')))

    return launch.LaunchDescription(includes)
```

#### Expected behavior
When accessing the environmental variable in node 'test_0' it should have the value '0'.
When accessing the environmental variable in node 'test_1' it should have the value '1'.
 

#### Actual behavior
When accessing the environmental variable in node 'test_0' it has the value '1'.
When accessing the environmental variable in node 'test_1' it has the value '1'.
 

#### Additional information

----

hidmic, 2021-12-07:
@fas2037 `SetEnvironmentVariable` mutates the environment of `launch` itself (which is one, later propagated to its children). Even if that wasn't the case, it still wouldn't do what you expect since `LaunchDescription` instances do not execute anything and there's no lexical scoping to rely on.

For your specific use case, I'd suggest using the `append_env` argument of the `Node` class. If you'd like to see scoping improved, there have been some discussions in https://github.com/ros2/launch/issues/501 and https://github.com/ros2/launch/issues/313.

jacobperron, 2022-03-04:
I _think_ https://github.com/ros2/launch/pull/601 resolves this ticket. @fas2037 correct me if I'm wrong.