ros2/launch #568: Cannot parse var in respawn

kosuke55, 2021-12-10:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 20.04
- Installation type:
  - both
- Version or commit hash:
  - ee9ee05abcfadd7ded03e455223c955538c118cd
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclcpp, rclpy

#### Steps to reproduce issue

Simple sample launch with respawn arg.
```
<launch>
  <arg name="respawn" default="false" />
  <node name="turtlesim" pkg="turtlesim" exec="turtlesim_node" output="screen" respawn="$(var respawn)" />
</launch>
```

Run following command.
```
ros2 launch turtle.launch
kill $(pgrep -f __node:=turtlesim)
```
The node respaws even though respawn is set to false.

#### Expected behavior
If false, do not respawn.
```
ros2 launch turtle.launch respawn:=false
kill $(pgrep -f __node:=turtlesim)
```

If true, respawn.
```
ros2 launch turtle.launch respawn:=true
kill $(pgrep -f __node:=turtlesim)
```

#### Actual behavior
Both will respawn. No matter what string we set, respawn works.
```
[INFO] [turtlesim_node-1]: process started with pid [3555054]
[turtlesim_node-1] 1639106343.701872 [100] turtlesim_: using network interface wlp0s20f3 (udp/192.168.50.218) selected arbitrarily from: wlp0s20f3, docker0
[turtlesim_node-1] [INFO 1639106343.708674157] [turtlesim]: Starting turtlesim with node name /turtlesim
[turtlesim_node-1] [INFO 1639106343.710621384] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2022-04-12:
I just ran into this while adding respawn support to Nav2. Its not yet on Rolling so just commenting here that its an issue but I expect that it will be removed in the next Rolling sync.

**Will this get in for Humble**? 

Also, the PR is merged, I think this ticket can be closed.

clalancette, 2022-04-13:
This has already been released into the launch packages in version 0.22.0.  Can you double-check that you are using at least that version?

kosuke55, 2022-04-25:
This issue was solved with  #569 and released in [0.22.0](https://github.com/ros2/launch/commits/0.22.0)

closed 2022-04-25 by kosuke55