ros2/launch #578: Cannot use event-triggered ReadyToTest

Aposhian, 2022-01-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.17.0
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
https://gist.github.com/Aposhian/cf1efc55311711649ad9c3b3e91f4c88

The following launch file is invalid, because the `ReadyToTest` action cannot be found through a recursive search.

```python
import unittest
import pytest

from launch import LaunchDescription
from launch.actions import ExecuteProcess, RegisterEventHandler
from launch.event_handlers import OnProcessExit
from launch_testing.actions import ReadyToTest

@pytest.mark.launch_test
def generate_test_description():
    setup = ExecuteProcess(cmd=["setup.sh"], output="screen")

    return (
        LaunchDescription(
            [
                setup,
                RegisterEventHandler(
                    OnProcessExit(target_action=setup, on_exit=ReadyToTest())
                ),
            ]
        ),
        {"setup": setup},
    )

class TestTeachAndRepeat(unittest.TestCase):
    def test_relying_on_setup(self, launch_service, proc_output):
        print("Running the test!")
```

#### Expected behavior
The tests run after appropriate stdout is detected from my setup script.

#### Actual behavior
```
launch_test.py: error: generate_test_description functions without a ready_fn argument must return a LaunchDescription containing a ReadyToTest action
```

#### Additional information
This makes `ReadyToTest` incredibly clunky: I can't think of a way to use it in a more intelligent way than with a hardcoded `TimerAction`, which is unreliable. I tried to use `ready_fn` for this use case, but it is also unusable because it has a 15 second hardcoded timeout, which is too short for my real use case.

This use-case could maybe be fixed if `OnActionEventBase` inherited from `EventHandler` instead of `BaseEventHandler`, so that it could actually have an implementation for `describe` that returns its sub-entities.

However, the current methodology will have no way of detecting more dynamic uses of `ReadyToTest`, like if you wanted to return it from a `Callable` in an `OnProcessIO` handler:

```python
OnProcessIO(
    target_action=setup,
    on_stdout=lambda stdout: ReadyToTest() if "done" in str(stdout) else None,
)
```

Aposhian, 2022-01-11:
If there are other ways to use `ReadyToTest` that circumvent the issues I have encountered, besides [immediately triggering the test like is done in the examples](https://github.com/ros2/launch/tree/master/launch_testing/test/launch_testing/examples), I would love to hear about it.

nnmm, 2022-09-26:
@Aposhian did you find out a better way to do this? I'm encountering the same problem.

Aposhian, 2022-09-26:
The workaround we used was just making our tests tolerant to have a large timeout in the things that they were testing. Not an ideal solution.

nnmm, 2022-09-26:
Thanks. @clalancette I think this should be labeled as a `bug` instead of a `question`, since this is a regression from `ready_fn`.

closed 2022-12-02 by methylDragon