ros2/launch #582: Make hardcoded 15 second launch_testing timeout configurable

Aposhian, 2022-01-13:
## Feature request

#### Feature description
Currently you only get [15 seconds](https://github.com/ros2/launch/blob/b59bb6b88b40eaa82870542d93bdfb1a033d05c3/launch_testing/launch_testing/test_runner.py#L184) to signal `ReadyToTest`. This is not configurable, which means that you can't use processes that take a long time to startup.

#### Implementation considerations
Ideally this could be configured at the launch file level. Maybe the tests themselves could say how long they are willing to wait for `ReadyToTest`?

clalancette, 2022-02-03:
Yeah, agreed that this would be nice to make configurable.  Please consider making a pull request to do so, we're happy to review.

jacobperron, 2022-06-15:
I also hit this hard-coded limit recently. It's easy to reproduce with the following launch test file:

```py
import launch
from launch.actions import TimerAction
import launch_testing.markers
from launch_testing.actions import ReadyToTest

def generate_test_description():
    return launch.LaunchDescription([
        TimerAction(period=16.0, actions=[ReadyToTest()]),
    ])
```

As a potential workaround, the alternative [`launch_pytest`](https://github.com/ros2/launch/tree/master/launch_pytest) package does not have such a limit. Here is the equivalent test file written with `launch_pytest`:

```py
import pytest

import launch
from launch.actions import TimerAction
import launch_pytest
from launch_pytest.actions import ReadyToTest

@launch_pytest.fixture
def launch_description():
    return launch.LaunchDescription([
        TimerAction(period=16.0, actions=[ReadyToTest()]),
    ])

@pytest.mark.launch(fixture=launch_description)
def test_case_1():
    pass
```

deepanshubansal01, 2022-07-11:
Closing the issue, since the feature has been implemented https://github.com/ros2/launch/pull/625

closed 2022-07-11 by deepanshubansal01