ros2/launch #588: Changing 'output' substitution type breaks dictionary usage

mjeronimo, 2022-01-24:
- Operating System: Ubuntu 20.04
- Installation type: source 
- Version or commit hash: 8006dfadcd4d77dfa0aedbc0030ae9ceab955165
- DDS implementation: any

#### Steps to reproduce issue
Save the following to bug.launch.py:
```Python
from launch import LaunchDescription
import launch_ros.actions

def generate_launch_description():
    return LaunchDescription([
        launch_ros.actions.Node(
            package='demo_nodes_cpp', executable='talker', output={
            'stdout': 'screen',
            'stderr': 'screen'
            }),
        launch_ros.actions.Node(
            package='demo_nodes_cpp', executable='listener', output='screen'),
    ])
```

Run with:
```
ros2 launch bug.launch.py
```
#### Expected behavior
The launch operation starts the talker and listener, with outputs directed to the screen.

#### Actual behavior
```
[INFO] [launch]: All log files can be found below /home/username/.ros/log/2022-01-24-08-22-15-307722-machinename-1831798
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): stdoutstderr is not a valid standard output config i.e. "screen", "log" or "both"
```

hidmic, 2022-01-26:
Ahh, I had forgotten about those options. @mjeronimo a follow-up PR to support both substitutions and dictionaries would be best and most welcome.

m-elwin, 2022-08-09:
Hi, I took the liberty to create a pull request for this issue. Happy to address any concerns since I'm not overall aware of the greater context of the changes that caused this issue in the first place. I can also work on a test but wasn't sure exactly where to begin with that.

mjeronimo, 2022-08-20:
Fixed by #640

closed 2022-08-20 by mjeronimo

FranekStark, 2022-09-27:
This issue is importunately not fixed in the current Humble release, are there any plans to do so?

FranekStark, 2022-10-03:
Today I figures out that the behavior of the output parameter in ros launch acts very unexpected.
Because of the issue described above i changed it to output=log, expecting that the rviz2 warnings will be send to the log file. However the INFO and WARNINGS logs still appeared in the terminal output. But when I then changed to output=own_log, the output in the terminal disappeared.
Anyone knows if this is another bug or is it related to this bug?