ros2/launch #604: Can't build launch_testing

Kaju-Bubanja, 2022-03-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - latest source
- Version or commit hash:
  - fbb5f514c4ae396ff9a12734ff2ab499032d59e4
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Trying to compile the repo with below command I get following error message:
```
colcon build --symlink-install --packages-select launch_testing
```
```
Starting >>> launch_testing
--- stderr: launch_testing                   
error: can't copy 'example_processes/terminating_proc': doesn't exist or not a regular file
---
Failed   <<< launch_testing [2.67s, exited with code 1]

```

#### Expected behavior
I would expect to be able to compile the repo
#### Actual behavior
See error above
#### Additional information

clalancette, 2022-03-30:
It seems to build fine here, with `colcon build --symlink-install --packages-up-to launch_testing`.  Can you try that with a clean workspace?

Kaju-Bubanja, 2022-04-13:
Trying the comment which you suggested: 
```
colcon build --symlink-install --packages-up-to launch_testing
Package 'launch_testing' specified with --packages-up-to was not found
```

even though I have it under src/ros2/launch/launch_testing and it is on the newest master commit. I have set this in my .bashrc: `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`. Is it possible that this is causing build problems. As far as I understand it shouldn't, but that's the only thing which I have in mind which is different than a totally clean install. Actually I tried to build without that export but I get the same results, but I still thought it can't hurt to mention it.

Kaju-Bubanja, 2022-04-14:
So sorry, I had a COLCON_IGNORE file in the launch_testing, don't know how I could have missed it.

closed 2022-04-14 by Kaju-Bubanja