ros2/launch #610: latest_test_result log:  the root tag is neither 'testsuite' nor 'testsuites'

AndyZe, 2022-04-11:
## Bug report

**Required Info:**

- Operating System:
  - ROS2 Rolling, latest binaries. I have seen the same behavior with Galactic binaries, too
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

Clone this "minimal example" repo and build it:  https://github.com/AndyZe/minimal_integration_test

Run tests:  `colcon test --packages-select minimal_integration_test`

Display the test results. You should see that all tests pass:  `colcon test-result --verbose`

Open `COLCON_WS/log/latest_test-result`. I see text like this:

```
...
[0.092s] Level 1:colcon.colcon_test_result.test_result:get_test_results(ctest)
[0.161s] Level 1:colcon.colcon_test_result.test_result:get_test_results(xunit)
[0.174s] DEBUG:colcon.colcon_test_result.test_result.xunit:Skipping 'build/minimal_integration_test/Testing/20220411-1703/Test.xml': the root tag is neither 'testsuite' nor 'testsuites'
```

#### Expected behavior

The log would show that the test passed.

#### Additional information

I have an answers.ros.org question here, but it's not stripped down to a minimal example:  [https://answers.ros.org/question/398825/colcon-test-passes-but-result-is-not-generated/](https://answers.ros.org/question/398825/colcon-test-passes-but-result-is-not-generated/)

hidmic, 2022-06-23:
@AndyZe I'm unable to reproduce on `master` (and tests fail):

<details>
<summary>CTest output</summary>

```
   Start 1: test_launch_basic_test.test.py

1: Test command: /usr/bin/python3.10 "-u" "/home/michel/Workspaces/jammy_ros2_ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test/test_results/minimal_integration_test/test_launch_basic_test.test.py.xunit.xml" "--package-name" "minimal_integration_test" "--output-file" "/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test/ros_test/test_launch_basic_test.test.py.txt" "--command" "ros2" "test" "/home/michel/Workspaces/jammy_ros2_ws/src/tmp/minimal_integration_test/test/launch/basic_test.test.py" "test_binary_dir:=/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test" "--junit-xml=/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test/test_results/minimal_integration_test/test_launch_basic_test.test.py.xunit.xml" "--package-name=minimal_integration_test"
1: Test timeout computed to be: 120
1: -- run_test.py: invoking following command in '/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test':
1:  - ros2 test /home/michel/Workspaces/jammy_ros2_ws/src/tmp/minimal_integration_test/test/launch/basic_test.test.py test_binary_dir:=/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test --junit-xml=/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test/test_results/minimal_integration_test/test_launch_basic_test.test.py.xunit.xml --package-name=minimal_integration_test
1: Running with ROS_DOMAIN_ID 1                        
1: ROS_DOMAIN_ID 1
1: [INFO] [launch]: All log files can be found below /home/michel/.ros/log/2022-06-23-20-00-21-550422-a00426f1dd45-210173
1: [INFO] [launch]: Default logging verbosity is set to INFO
1: test_gtest_run_complete (minimal_integration_test.TestGTestWaitForCompletion) ... Warning: Automatically adding attributes like self.proc_info to the test class will be deprecated in a future release.  Instead, add proc_info to the test method argument list to access the test object you need
1: [INFO] [basic_test-1]: process started with pid [210175]
1: [basic_test-1] [==========] Running 1 test from 1 test suite.
1: [basic_test-1] [----------] Global test environment set-up.
1: [basic_test-1] [----------] 1 test from TaskPlanningFixture
1: [basic_test-1] [ RUN      ] TaskPlanningFixture.TrueIsTrueTest
1: [basic_test-1] [ERROR] [1656014421.590486880] [basic_test]: DONE WITH CONSTRUCTOR!!
1: [basic_test-1] [ERROR] [1656014421.590580674] [basic_test]: DONE WITH SETUP!!
1: [basic_test-1] TEST BEGINNING!!
1: [basic_test-1] DONE WITH TEARDOWN
1: [basic_test-1] [       OK ] TaskPlanningFixture.TrueIsTrueTest (16 ms)
1: [basic_test-1] [----------] 1 test from TaskPlanningFixture (16 ms total)
1: [basic_test-1] 
1: [basic_test-1] [----------] Global test environment tear-down
1: [basic_test-1] [==========] 1 test from 1 test suite ran. (16 ms total)
1: [basic_test-1] [  PASSED  ] 1 test.
1: [basic_test-1] DONE SHUTTING DOWN ROS
1: [INFO] [basic_test-1]: process has finished cleanly [pid 210175]
1: ok
1: Processes under test stopped before tests completed
1: 
1: ----------------------------------------------------------------------
1: Ran 1 test in 0.139s
1: 
1: OK
1: -- run_test.py: return code 1                       
1: -- run_test.py: verify result file '/home/michel/Workspaces/jammy_ros2_ws/build/minimal_integration_test/test_results/minimal_integration_test/test_launch_basic_test.test.py.xunit.xml'
1/4 Test #1: test_launch_basic_test.test.py ...***Failed    0.58 sec
```

</details>

griswaldbrooks, 2023-09-09:
@AndyZe had migrated his code to this repo
https://github.com/PickNikRobotics/ros_testing_templates/tree/ros2/minimal_integration_test
and I think that
https://github.com/PickNikRobotics/ros_testing_templates/blob/ros2/minimal_integration_test/test/launch/basic_test.test.py#L32
may have fixed the issue, if it's the came as in the ros answer

clalancette, 2023-09-13:
All right, given the last comment I'm going to close this out.  Feel free to reopen if that is in error.

closed 2023-09-13 by clalancette