ros2/launch #618: $(dirname) is affected by previous ($find-pkg-share ...)

felixdivo, 2022-05-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04...
- Installation type:
  - binaries
- Version or commit hash:
  - up to date
- DDS implementation:
  - irrelevant
- Client library (if applicable):
  - irrelevant

#### Steps to reproduce issue

Create file `test.launch.yaml`.

```yaml
launch:
  - executable:
      cmd: echo $(dirname)
      name: obvious
      output: screen
  - include:
      file: $(find-pkg-share some_package)/file.launch.yaml
  - executable:
      cmd: echo $(dirname)
      name: surprise
      output: screen
```

The file `file.launch.yaml` in `some_package`:
```yaml
launch: []
```

Run `ros2 launch test.launch.yaml`.

#### Expected behavior

`echo $(dirname)` outputs the path to `test.launch.yaml` each time.

#### Actual behavior

`echo $(dirname)` outputs the path to `test.launch.yaml` the first time and the path to the package share of `some_package` the second time. It appears that "the context" is somehow not reset. Is this really intentional? I also could not find the docs for that substitution.

#### Additional information

```yaml
launch:
  - executable:
      cmd: echo $(dirname)
      name: obvious
      output: screen
  - executable:
      cmd: echo $(find-pkg-share some_package)
      name: testing_this
      output: screen
  - executable:
      cmd: echo $(dirname)
      name: surprise
      output: screen
```

In this setting `echo $(dirname)` outputs the path to `test.launch.yaml` each time.

#### Implementation considerations

It is very unexpected but also breaking when changed.

jacobperron, 2022-06-13:
I can confirm the reported behavior; the same applies to a regular Python launch file:

```py
import launch
from launch.actions import ExecuteProcess
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import ThisLaunchFile

from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    return launch.LaunchDescription([
        ExecuteProcess(name='obvious', cmd=['echo', ThisLaunchFile()], output='screen'),
        IncludeLaunchDescription(PythonLaunchDescriptionSource([FindPackageShare('dummy_robot_bringup'), '/launch/dummy_robot_bringup.launch.py'])),
        ExecuteProcess(name='surprise', cmd=['echo', ThisLaunchFile()], output='screen'),
    ])
```

I've marked this ticket as a bug since I don't think this is the intended behavior for the `ThisLaunchFile` substitution.

jacobperron, 2022-06-13:
So, the issue happens because `ThisLaunchFile` returns the value from the launch context's locals,

https://github.com/ros2/launch/blob/074cd2903ddccd61bce8f40a0f58da0b7c200481/launch/launch/substitutions/this_launch_file.py#L59

and this value is overridden by `IncludeLaunchDescription` when it is executed,

https://github.com/ros2/launch/blob/074cd2903ddccd61bce8f40a0f58da0b7c200481/launch/launch/actions/include_launch_description.py#L150-L152

Note, I think we have the same issue with [`ThisLaunchFileDir`](https://github.com/ros2/launch/blob/074cd2903ddccd61bce8f40a0f58da0b7c200481/launch/launch/substitutions/this_launch_file_dir.py).

One possible solution that comes to mind is to scope the context locals to the `IncludeLaunchDescription` action. We could achieve this by changing [this line](https://github.com/ros2/launch/blob/074cd2903ddccd61bce8f40a0f58da0b7c200481/launch/launch/actions/include_launch_description.py#L185) to something like,

```py
return [PushContextLocals(), *set_launch_configuration_actions, launch_description, PopContextLocals()]
```

where `PushContextLocals()` and `PopContextLocals()` are new actions. If we don't want to expose these are new actions, we could use an `OpaqueFunction`.

The question I have is whether this will disrupt other functionality. I don't actually know the intention behind `context.locals`, so scoping them to an included description might have undesired side-effects. @wjwwood @hidmic @ivanpauno Thoughts?

fnobis, 2022-10-06:
I am facing the same issue. Is there any workaround for this, until it is fixed on rolling?

icolwell-as, 2025-03-13:
I ran into this bug today using XML launch files. 

> Is there any workaround for this, until it is fixed on rolling?

I was able to work around it by adding `<let name="dirname" value="$(dirname)"/>` to the very top of the launch file, then using `$(var dirname)` in place of `$(dirname)` later in the file.

peci1, 2025-03-30:
Wow. Such a bummer. Another thing that makes the transition to ROS 2 a real pain.

I've started writing down a list: https://gist.github.com/peci1/52aea17faf5f7e33c8096b75dc9bcd2f . When I have time, I might get to working on some of the listed issues :)

christophebedard, 2025-04-23:
FYI I opened a PR to fix this issue: #862. If anyone who hit this issue could give it a try and let me know if it works for you, that would be appreciated!