ros2/launch #620: $(arg) are overwritten when including multiple launch files

h-ohta, 2022-06-16:
## Bug report ?

**Required Info:**

- Operating System:
  - Ubuntu20.04
- Installation type:
  - galactic binaries 
- Version or commit hash:
  - up to date
- DDS implementation:
  - irrelevant
- Client library (if applicable):
  - irrelevant

#### Steps to reproduce issue

- Create three files
```
# outside.launch.xml

<launch>
    <include file="inside1.launch.xml"></include>
    <include file="inside2.launch.xml"></include>
</launch>

#include1.launch.xml

<launch>
    <arg name="test_arg" default="1"/>
    <log message="include1 has $(var test_arg)"/>
</launch>

#include2.launch.xml

<launch>
    <arg name="test_arg" default="2"/>
    <log message="include2 has $(var test_arg)"/>
</launch>
```

- Run `outside.launch.xml`

#### Expected behavior

```bash
$ ros2 launch outside.launch.xml 
[INFO] [launch]: All log files can be found below /home/hrkota/.ros/log/2022-06-16-14-28-00-475031-npc2103034-543336
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: include1 has 1
[INFO] [launch.user]: include2 has 2
```
#### Actual behavior

```bash
$ ros2 launch outside.launch.xml 
[INFO] [launch]: All log files can be found below /home/hrkota/.ros/log/2022-06-16-14-28-00-475031-npc2103034-543336
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: include1 has 1
[INFO] [launch.user]: include2 has 1
```

#### Additional information

Now, I can avoid this problem by using <group> tag.

```
# outside.launch.xml

<launch>
    <group>
        <include file="inside1.launch.xml"></include>
    </group>

    <group>
        <include file="inside2.launch.xml"></include>
    </group>
</launch>

# result

$ ros2 launch outside.launch.xml 
[INFO] [launch]: All log files can be found below /home/hrkota/.ros/log/2022-06-16-14-30-35-485185-npc2103034-543607
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: include1 has 1
[INFO] [launch.user]: include2 has 2

```

Is it bug or correct behavior?
I can't get some information about it.

----
## Feature request

#### Feature description

#### Implementation considerations

h-ohta, 2022-06-21:
@clalancette Could you confirm this? Thank you.

hidmic, 2022-06-22:
@h-ohta `include` actions don't scope (almost no action does), only [`group`](https://github.com/ros2/launch/blob/074cd2903ddccd61bce8f40a0f58da0b7c200481/launch/launch/actions/group_action.py#L43-L47) actions do.

peci1, 2025-03-30:
This is really a hard hit when migrating from ROS 1. I know it is described in the migration guide, but for people who don't know ROS 2 well (yet), the formulation does not seem to be warning them enough:

> Available in ROS 1, included content was scoped. In ROS 2, it’s not. Nest includes in group tags to scope them.

This basically means that migrating ROS 1 launch files into ROS 2 is `<include>` -> `<group><include>`. What about providing an `<include scoped="1">` attribute or something similar that would ease that transition?

christophebedard, 2025-03-30:
> What about providing an `<include scoped="1">` attribute or something similar that would ease that transition?

See #801 and #841