ros2/launch #635: Keyboard input fails to be processed when node is launched by launch.py file

Curiosi-Orch, 2022-07-29:
## Bug report

Keyboard input fails to be processed when node is launched by launch.py file

- Operating System:
Ubuntu 18.04
- Installation type:
ROS2 dashing 
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue
c++ getchar() function cannot be processed when the rclcpp node is launched in launch.py file, and neither was the case when I launched the turtle_teleop_key node in turtlesim tutorial package in the same way. I am not sure if there should be any modification for the launch.py file. btw, it works well in ROS1.

#### Expected behavior
program can read the keyboard input correctly.

#### Actual behavior
no response to the keyboard input.

#### My code in test.launch.py
def generate_launch_description():
  test = Node(package='turtlesim',
              node_name='turtle_teleop_key',
              node_executable='turtle_teleop_key',
              output='screen')      
  return LaunchDescription([test,])

clalancette, 2022-07-29:
> c++ getchar() function cannot be processed when the rclcpp node is launched in launch.py file, and neither was the case when I launched the turtle_teleop_key node in turtlesim tutorial package in [...]

In my opinion, this is a feature, not a bug.  A launch file is meant to launch many processes; what happens if you have 2 (or more) of them that both want to consume stdin?  Which one should get it?

That said, I'll leave this open for a little while to see if anyone has other ideas, or a straightforward way to make this determination (and make it clear to users).

Curiosi-Orch, 2022-07-31:
> In my opinion, this is a feature, not a bug. A launch file is meant to launch many processes; what happens if you have 2 (or more) of them that both want to consume stdin? Which one should get it? [...]

Thank you for your reply! it makes sense, but also makes some inconvenience. lol

clalancette, 2022-08-18:
I'm going to close this one out, as I don't think there is much to do here.  If you disagree, feel free to reopen (but I would also say that we would need an idea on how we would possibly fix this to consider it).

closed 2022-08-18 by clalancette