ros2/launch #645: "emulate_tty" not implemented in YAML or XML launch files

ciandonovan, 2022-08-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Run the following launch files:

```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():

    node = Node
    (
        package='robot_gui',
        executable='robot_gui',
        name='robot_gui',
        output='screen',
        emulate_tty='True',
    )

    return LaunchDescription
    ([
        node,
    ])
```
```
<launch>
        <node pkg="robot_gui" exec="robot_gui" name="robot_gui" output="screen" emulate_tty="true" />
</launch>
```
```
---
launch:
  - node:
      pkg: "robot_gui"
      exec: "robot_gui"
      name: "robot_gui"
      output: "screen"
      emulate_tty: true
```

#### Expected behavior

Standard Output is properly displayed.

#### Actual behavior

Some Standard Output missing when `emulate_tty` is not specified. When it is specified...

Error with XML launch file:
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [xml]: Unexpected attribute(s) found in `node`: {'emulate_tty'}
```

Error with YAML launch file:
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [yml]: Unexpected key(s) found in `node`: {'emulate_tty'}
```

#### Comment

While the ROS2 documentation states that some lower-level functionality may only be accessible through Python, I feel that the ability to reliably display Standard Output should be included as a common feature across all launch file types.

Given that many users will be migrating from ROS1 where they were familiar with XML type launch files and might feel be inclined to continue to use them, this undocumented lack of functionality is very likely to trip them up.

mcullinan11, 2022-09-13:
I ran into this issue too. I definitely agree about the importance of maintaining this functionality in XML launch files

adityapande-1995, 2023-06-08:
Closing this, as this has been implemented : https://github.com/ros2/launch/pull/669

closed 2023-06-08 by adityapande-1995