ros2/launch #650: Request for Release in Humble and Rolling

ijnek, 2022-09-11:
With #640 and #647 (backport PR) merged, may I request the maintainers to make a release for Rolling and Humble please?
It fixes some unintended breakages caused by #588, and I believe it affects a fair number of projects. Thanks!

clalancette, 2022-09-13:
This is done in https://github.com/ros/rosdistro/pull/34520 (and I'll be merging it soon), so closing this out.

closed 2022-09-13 by clalancette

ijnek, 2022-09-13:
Thank you @clalancette !

stefanscherzinger, 2022-09-25:
May I ask if this gets released for Humble as well? The respective PR is not yet listed [the rosdistro's PRs](https://github.com/ros/rosdistro/pulls?q=is%3Apr+launch+humble+in%3Atitle+) and the `launch` version still seems to be [1.0.2-1](https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L1869).

FranekStark, 2022-09-29:
@stefanscherzinger I agree to this. The bug is really annoying and should be fixed in humble asap.

stefanscherzinger, 2022-09-29:
@FranekStark 

In the meantime, I'm using some practical hack for my launch files:

```python
import os
if os.environ['ROS_DISTRO'] == 'humble':
    output = "both"
else:  # foxy, galactic
    output = {"stdout": "screen", "stderr": "screen"}
```
and then
```bash
    my_node = Node(
        package="some_package",
        executable="some_executable",
        output=output
        # ...
    )
```