ros2/launch #658: Expose the RegisterEventHandler action

esteve, 2022-10-04:
## Feature request

#### Feature description

Based on the discussion in https://github.com/ros2/launch_ros/issues/236#issuecomment-849975470, it'd be good to expose the RegisterEventHandler action so as to be able to support LifecycleNodes in launch_ros

esteve, 2022-10-04:
@jacobperron @kenji-miyake for an example Python code like this:

```python
        RegisterEventHandler(
            OnProcessStart(
                target_action=turtlesim_node,
                on_start=[
                    LogInfo(msg='Turtlesim started, spawning turtle'),
                    spawn_turtle
                ]
            )
        ),
```

I thought the XML could look like as follows:

```xml

<register_event_handler>
  <event_handler name="on_process_start">
    <target_action>turtle_sim_node<target_action/>
    <arg name="on_start">
        <action>log_info</action>
        <action>spawn_turtle</action>
    </arg>
  </event_handler>
</register_event_handler>
```

But I'd appreciate any feedback. Thanks.

andreas-loeffler, 2026-03-13:
Hi @esteve :)  I'm running into this exact limitation while trying to migrate some launch files to xml. Are there any updates or active plans regarding this Issue or the Draft PR #659?