ros2/launch #691: High memory usage on log burst

kamibo, 2023-02-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries (apt)
- Version or commit hash:
  - Humble
- DDS implementation:
  - FastDDS

#### Steps to reproduce issue

Launch one or many nodes that logs a lot (using functions such as `RCLCPP_INFO` and a 1ms timer for instance).

#### Expected behavior

`ros2 launch` should be stable regarding regarding memory usage.

#### Actual behavior

`ros2 launch` keeps on consuming memory.

#### Additional information

`launch_service` queue (`asyncio.Queue()`) has no limit.
And only one event is handled per loop cycle (`this_loop.create_task(self._process_one_event())`).
So the queue keeps growing until no memory left.

Note that trying to handle many events in `_process_one_event` seems to solve the issue.

ZhenshengLee, 2023-11-14:
@kamibo Thanks for your report!

I also noticed about the heavy memory usage of `launch.py`, in Foxy version.

Could you tell me which command you use to monitor the memory usage of ros2 nodes or processes? Any recommendation tool besides `top`?

Thanks.

kamibo, 2023-11-14:
@ZhenshengLee  Actually I've used `htop`.

HyeKaguya233, 2023-12-15:
Could you share the method you used to solve this?I also had this problem.

HyeKaguya233, 2023-12-16:
i will try,thx!!