ros2/launch #727: Parse arguments into their relevant type

nlamprian, 2023-08-31:
## Feature request

#### Feature description
Substitutions are meant to provide values for a dynamic launch configuration and thus they evaluate to strings. Substitutions though are also used to execute code (see PythonExpression). When that's performed, the result of a substitution must be wrapped in quotes as it needs to be defined as a string, and then it is compared to other strings. For example

```python
PythonExpression(["'value1' if '", LaunchConfiguration("name"), "' == 'true' else 'value2'"])
```

That's a primitive and cumbersome way of working with arguments. It would be useful if it was possible for the launch configurations to be parsed into the relevant type of their argument.

#### Implementation considerations
We could hopefully have specialized launch configurations, `{Bool,Int,Float,etc}LaunchConfiguration`, that parse their value before returning it. Then, an expression like the above would become

```python
PythonExpression(["'value1' if ", BoolLaunchConfiguration("name"), " else 'value2'"])
```

Do you think that's reasonable and would you be interested in a PR?

sloretz, 2023-09-07:
Is this the feature Idea something like `BooLaunchConfiguration("my_arg")` takes yaml true/false (ex: `my_arg:=true` or `my_arg:=no`), and returns a string that evaluates to Python True/False (ex: `"True"` or `"False"`) such that when used in `PythonExpression()` it can be included in a substitution without wrapping it in quotes or a type cast?

There are some substitutions that [convert to yaml boolean strings](https://github.com/ros2/launch/blob/rolling/launch/launch/substitutions/boolean_substitution.py#L205-L210
), so making launch configurations that returned Python True/False strings would depart from yaml types being used.

 Is there a way we could get the behavior without `PythonExpression`? I think it's ok if using  `PythonExpression` is a little clunky as it's there to make all cases possible, but common cases should have a more specific substitution.

What about a substitution that returned different values abased on the yaml true-ness of the argument, such that `my_arg:=true` resulted in the substitution returning either `"value1"` or `"value2"`
?

```Python
IfThenElseSubstitution(if_=LaunchConfiguration("my_arg"), then="value1", else_="value2")
```

nlamprian, 2023-09-07:
Although I have only come across boolean cases in my work so far, I would still hope to have typed substitutions so that I can directly compare any type. But yes, ultimately, the goal would be to provide such a conditional substitution, possibly in both the `if` and `unless` form, where you specify the first value and have the second one default to an empty string.

clalancette, 2023-09-12:
We discussed this in the weekly meeting, and since we only have a known use case for booleans for now, we think we should stick to that.

So we think we should go with @sloretz's idea of a `IfThenElseSubstitution`; the goal should always be to get away from `PythonExpression` as much as possible.  @nlamprian if you are good with that, we'd be happy to review a PR implementing it.

PR #734 (2023-09-14): Add conditional substitution
Closes #727
commit 11c0b978 (2023-09-14): Add conditional substitution (#734)

* Add conditional substitution

Closes: #727
commit 159ee9cc (2023-09-20): Add conditional substitution (#734)

* Add conditional substitution

Closes: #727
commit 2ee9e321 (2024-05-08): Add conditional substitution (#734) (#776)

* Add conditional substitution

Closes: #727
closed 2023-09-14 by clalancette