ros2/launch #733: Node in launch appends empty "--ros-args" generating segmentation fault

muttistefano, 2023-09-12:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22 , ros rolling from binaries

Hi all,
up to few days ago, the following launch snippet was working fine:

```
    spawn_sweepee_1 = Node(
        package='ros_gz_sim',
        executable='create',
        output='screen',
        arguments=['-file', sweepee_1_path,
                   '-name', 'sweepee_1',
                   '-allow_renaming', 'false',
                   '-x', '-3.5',
                   '-y', '2.2',
                   '-z', '0.3']
    )
```

and it was creating the following command:

```
/home/gino/ROS/lampo_ws_ros2_rolling/install/ros_gz_sim/lib/ros_gz_sim/create -file /home/gino/ROS/lampo_ws_ros2_rolling/install/lampo_description/share/lampo_description/urdf/diff1.sdf -name sweepee_1 -allow_renaming false -x -3.5 -y 2.2 -z 0.3
```
After a recent update, the generated command is the same, with the `--ros-args` appended : 

```
/home/gino/ROS/lampo_ws_ros2_rolling/install/ros_gz_sim/lib/ros_gz_sim/create -file /home/gino/ROS/lampo_ws_ros2_rolling/install/lampo_description/share/lampo_description/urdf/diff1.sdf -name sweepee_1 -allow_renaming false -x -3.5 -y 2.2 -z 0.3 --ros-args
```

this line generates a segmentation fault.

I noticed that if i set the node name inside the Node name, something gets appended after  `--ros-args` , but is not coherent with rcl:
```
This syntax is deprecated. Use '--ros-args --remap __node:=gino' instead
```

Is there a fix to this ? 
Thanks

clalancette, 2023-09-12:
So I can reproduce the problem where it is adding a `--ros-args` with no arguments on the end; we can look into that.

But I cannot reproduce the problem with the node crashing, nor with it giving the "deprecated" warning message.  I'm not sure what is going on there.  It would help to have a fully reproducible example here.

clalancette, 2023-09-21:
After talking about this a bit, there may be some issues in ros_gz_sim on how it parses command-line arguments.  However, I'm still going to wait to hear back before we move it around, to see if we can reproduce the crash here.

mjcarroll, 2023-09-21:
> After talking about this a bit, there may be some issues in ros_gz_sim on how it parses command-line arguments. 

https://github.com/gazebosim/ros_gz/pull/453

muttistefano, 2024-01-19:
i guess is fixed :)

closed 2024-01-19 by muttistefano