ros2/launch #741: Substitutions for string manipulations and xacro

nlamprian, 2023-11-04:
## Feature request

#### Feature description
Coming back with another request for a substitution... I tried using a path join substitution with a filename that depends on a launch configuration. The path join substitution does not allow one to concatenate parts of a filename. It ended up using a python expression.

```python
xacro_filename = PythonExpression(["'", LaunchConfiguration("model"), "' + '.xacro'"])
xacro_filepath = PathJoinSubstitution(["robot_description", "urdf", xacro_filename])
```

#### Implementation considerations
I would have hoped that text substitution could take multiple texts so that I could do

```python
TextSubstitution(LaunchConfiguration("model"), ".xacro")
```

But this is not compatible, given how text substitution was put together.

So, how about having a substitution that concatenates strings and then also having the separator parameterized?

In the simple case, one would do

```python
StringSubstitution([LaunchConfiguration("model"), ".xacro"])  # -> "model_value.xacro"
```

And then, this could be used in other practical ways, for example, to create a xacro substitution

```python
class XacroSubstitution(Command):
    def __init__(
        self, 
        filepath: SomeSubstitutionsType, 
        arg_str: SomeSubstitutionsType = "",  # xacro arguments that are passed as a launch argument
        arg_dict: Dict = None,  # xacro arguments that are defined in the launch file
    ):
        command = [FindExecutable(name="xacro"), " ", filepath, " "]
        arg_str, arg_dict = normalize_to_list_of_substitutions(arg_str), arg_dict or {}
        arg_list = [StringSubstitution([key, value], sep=":=") for key, value in arg_dict.items()]
        command.extend(StringSubstitution(arg_list + arg_str, sep=" "))
        super().__init__(command)
```

matosinho, 2024-01-29:
I did the same with Python expression for ".sdf". That was the exact thing I tried, to use a list with TextSubstitution, but then I discovered that it was not possible. Nice suggestion!

erichlf, 2024-03-13:
I attempted to setup a launch file for gazebo and I was hoping to set the environment variable needed for the model and world by using this same method. Of course, I can't because I can't convert the `PathJoinSubstitution` to a string.

christophebedard, 2025-03-16:
I wonder if we could make [`SomeSubstitutionsType`](https://github.com/ros2/launch/blob/151b024a7cfac06f0d56798cbe87ab180b9ea3ab/launch/launch/some_substitutions_type.py#L24-L28) accept lists of lists of text/substitutions, similar to what @dvdmc suggested in https://github.com/ros2/launch/issues/835#issuecomment-2727597475. Something like:

```py
SomeSubstitutionsType = Union[
    Text,
    Substitution,
    Iterable[Union[Text, Substitution, Iterable[Text, Substitution]]],
]
```

I don't know if this is doable in practice or if it has any unforeseen implications. If anyone could give it a try and find out, it would be appreciated!

christophebedard, 2025-03-16:
Wait, `PathJoinSubstitution` currently takes in a `Iterable[Union[Text, Substitution]]`: https://github.com/ros2/launch/blob/151b024a7cfac06f0d56798cbe87ab180b9ea3ab/launch/launch/substitutions/path_join_substitution.py#L29

If it took in an `Iterable[SomeSubstitutionsType]`, then the example in https://github.com/ros2/launch/issues/835#issuecomment-2727597475 might be possible. That might be the simplest solution for `PathJoinSubstitution`.

christophebedard, 2025-03-16:
@dvdmc @nlamprian @matosinho: please see #838. It improves `PathJoinSubstitution` to allow something like:

```py
PathJoinSubstitution(['robot_description', 'urdf', [LaunchConfiguration('model'), '.xacro']])
```

PR #838 (2025-04-15): Allow concatenating each path component of PathJoinSubstitution
Closes #741

As mentioned in the above issue, users want to concatenate multiple substitutions into a single string, which, in this case, is a single path component. This is a very normal use of substitutions, but `PathJoinSubstitution` does not support it. It only supports joining individual substitutions into a path.

This PR changes `PathJoinSubstitution` to accept an `Iterable[SomeSubstitutionsType]` instead of being restricted to `Iterable[Union[Text, Substitution]]`. This enables the use-case presented in the issue:

```py
P [... cut]
closed 2025-04-15 by christophebedard